armbian/build #14: u-boot broken

driv3l, 2015-02-03:
Igor,

I was trying out your debian build for BananaPi and ran into an issue. My initial build worked fine, but for some reason, I had to delete my local files and rebuild the image. The latest build does not appear to be working. u-boot cannot find the kernel. It tries to boot using TFTP instead of booting off SD card. I noticed you made some changes recently to support Orange Pi. I was wondering if that might have broken something.

igorpecovnik, 2015-02-03:
Hi,

Latest commit didn't touch that part so it must be something wrong with u-boot source. It happened before and will happen in the future. Build script is always updating to source when you start it. It's an advantage and disadvantage as well.

I will take a close look ASAP. Anyway I have to rebuild images. There are some kernel patches waiting to be implemented.

driv3l, 2015-02-03:
Thanks for the quick response. Hopefully you can get to the bottom of it since I am a little lost trying to find the problem.

igorpecovnik, 2015-02-04:
I have to wait that guys from u-boot fix it. If you are in a rush get an older commit.

closed 2015-02-04 by igorpecovnik

digitalhack, 2015-02-07:
It looks like the problem with u-boot is as follows:

Line 196 of u-boot/include/config_distro_bootcmd.h "part list ${devtype} ${devnum} devplist; " will cause an error unless u-boot is compiled with CONFIG_PARTITION_UUIDS and CONFIG_PART_CMD defined. This appears to have been introduced in the patch "sjoerd-ccu authored on Jan 5".

For the sunxi platform these are not defined hence the error.  I found I was able to fix it by adding at line 102 in uboot/include/configs/sunxi-common:

#define CONFIG_PARTITION_UUIDS
#define CONFIG_CMD_PART

There may be a better way to do this as I am no expert.

Hope this helps,

digitalhack

driv3l, 2015-02-07:
Great... thanks for spending the time to look into it. 

I will try out your fix later today.