armbian/build #25: DTB files in sub directorys will produce unknown machine-id error

db260179, 2015-03-13:
Very odd issue I have discovered?

Using the mainline kernel 3.19 for cubieboard2 and cubietruck, the u-boot has an issue with detecting the correct machine-id when the dtb file is located in /boot/dtb/ using the variable {fdtfile}.

Specifying the dtb name works fine, but using a variable in a sub folder will not detect the machine-id?

/config/boot-next.cmd:ext4load mmc 0 0x49000000 /boot/dtb/3.19-kernel/${fdtfile} - fail

/config/boot-next.cmd:ext4load mmc 0 0x49000000 /boot/${fdtfile} - works?

Any ideas? any patches you have applied but forgot to add to this github repo?

Thanks
David

igorpecovnik, 2015-03-17:
Huh, probably uboot issue. Haven't notice. Recently I changed this part anyway. I am packing DTB files separate and they are now simply under /boot/dtb/ and no more under /boot/dtb/x.xx-kernel/

db260179, 2015-03-21:
OK, after much debugging - There appears to be an issue with the 2015.1 branch of u-boot

It appears not to be passing the correct mach id to the kernel causing it to not boot.

Using 2014.10 works without this issue

git checkout v2014.10

igorpecovnik, 2015-03-21:
Thanks for update.

closed 2015-03-21 by igorpecovnik