armbian/build #28: please enable CONFIG_NETFILTER_XT_TARGET_LOG as a module

akriegisch, 2015-03-19:
this is required for ufw to work correctly. Please see https://bugs.launchpad.net/ufw/+bug/1294544 for details; seems there are some people with embedded systems trying to use ufw.

hvenzke, 2015-03-19:
hello Andreas  and others 

I can confirm this for the banana-pi-debian-sd-image creted by Igor.

and again performane matters , but without _required_ CORE  kernel modules this performace value are ZERO

<pre>
oot@bananapi ~ # diff rasberian bananapi
12d11
< CONFIG_NETFILTER_XT_TARGET_AUDIT=m
15a15
> CONFIG_NETFILTER_XT_TARGET_CONNSECMARK=m
19d18
< CONFIG_NETFILTER_XT_TARGET_HMARK=m
22d20
< CONFIG_NETFILTER_XT_TARGET_LOG=m
33a32
> CONFIG_NETFILTER_XT_TARGET_SECMARK=m
37d35
< CONFIG_NETFILTER_XT_MATCH_BPF=m
41d38
< CONFIG_NETFILTER_XT_MATCH_CONNLABEL=m
</pre>

akriegisch, 2015-03-20:
This has been fixed by https://github.com/igorpecovnik/lib/commit/02794be9cb0c0dbe18e93e98a3d1ce1e293b368f
Thanks, Igor!

closed 2015-03-20 by akriegisch