armbian/build #29: please enable CONFIG_CHR_DEV_SG as a module

akriegisch, 2015-03-19:
I am not sure if there is a more generic use case; I personally need SCSI generic devices for mode switching support of some usb sticks and it would be great to get that support in your kernel package!

hvenzke, 2015-03-19:
I can confirm that Igors Images lacs the USB support near compllte

Rasberian 
<pre>
Rasberian :

rnot@fw1 ~ $ zcat /proc/config.gz | egrep '(USB)' | egrep -v '(#)' | wc -l
292

bannanaPI Image from igor 
rnot@fw1 ~ $ 
 # zcat /proc/config.gz | egrep '(USB)' | egrep -v '(#)' | wc -l
29
</pre>
 
My Advise for Igor : 

Please use the SAME BASE kernel config as raspberrypi/linux
OR Just JOIN raspberrypi/linux with your code as its dublicate work for the same.

igorpecovnik, 2015-03-19:
>Please use the SAME BASE kernel config as raspberrypi/linux

It won't work / help without extended merging and code adjustment which I don't have resources for.

>I can confirm that Igors Images lacs the USB support near completely

Why there might not be USB?
http://linux-sunxi.org/Linux_mainlining_effort

igorpecovnik, 2015-03-19:
Kernel recompiled with https://github.com/igorpecovnik/lib/commit/02794be9cb0c0dbe18e93e98a3d1ce1e293b368f

deb package
http://mirror.igorpecovnik.com/kernel/3.19.2-bananapi-next.tar

closed 2015-03-19 by igorpecovnik