armbian/build #33: agetty + systemd + debian = serial login fail.

npotts, 2015-03-25:
Systemd, agetty and an embedded system do not seem to play nice.  Using the default settings will spawn a terminal on /dev/ttyS0 at 9600 baud, not the selected 115200.  Not sure how to fix this as it is coming from upstream, so your options are

1. Once your Olimex Lime/Lime2/Micro has booted, Switch the baud rate to 9600, hit enter and login as normal.
1.  Fix the root problem temporarily via something like this

    ```sh
        cat /lib/systemd/system/serial-getty@.service | sed s/"--keep-baud 115200,38400,9600"/"-L 115200"/g > /etc/systemd/system/serial-getty@ttyS0.service
        ln -s /etc/systemd/system/serial-getty@ttyS0.service /etc/systemd/system/getty.target.wants/
        systemctl daemon-reload
        systemctl start serial-getty@ttyS0.service
    ```

This should probably be marked as wontfix or simliar, but just in case someone else is using your distributions, it would be good to note this if you need a serial terminal.

igorpecovnik, 2015-03-26:
I noticed this but don't have a proper solution yet. It's noted under "known bugs" on my blog. Thanks for bringing this up, maybe somebody has a solution?

npotts, 2015-04-01:
Leme poke around in your codebase so see if I can insert a few hack lines to help with that

npotts, 2015-04-09:
This should be fixed in https://github.com/igorpecovnik/lib/pull/47

igorpecovnik, 2015-05-02:
It's working fine now, thanks!

closed 2015-05-02 by igorpecovnik