armbian/build #34: Mainline Errors

blindpet, 2015-03-27:
Trying to compile mainline 4.0 with port multiplier support.
All I did was edit the build.sh to say BRANCH="next"

Near the end I get these errors

make[5]: *** [drivers/net/phy/swconfig.o] Error 1
make[4]: *** [drivers/net/phy] Error 2
make[3]: *** [drivers/net] Error 2
make[2]: *** [drivers] Error 2
make[1]: *** [deb-pkg] Error 2
make: *** [deb-pkg] Error 2
tar: *.deb: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
rm: cannot remove ‘*.deb’: No such file or directory
------ Install kernel
dpkg: error processing /tmp/*image*.deb (--install):
 cannot access archive: No such file or directory
Errors were encountered while processing:
 /tmp/*image*.deb
dpkg: error processing /tmp/*headers*.deb (--install):
 cannot access archive: No such file or directory
Errors were encountered while processing:
 /tmp/*headers*.deb
dpkg: error processing /tmp/*dtb*.deb (--install):
 cannot access archive: No such file or directory
Errors were encountered while processing:
 /tmp/*dtb*.deb

igorpecovnik, 2015-03-27:
There are two possible scenarios
- patches are not compatible with 4.x, use branch 3.19.3 (this one works 100%)
- sometimes kernel clean helps

blindpet, 2015-03-27:
Ok, retrying with BRANCH="3.19.3" right?

How do I kernel clean exactly?

Also is it possible to add some apt-get install commands in build.sh?

igorpecovnik, 2015-03-27:
there is a parameter in my script: kernel_clean="yes" (now is no i guess).

I will add this functionality in next commit, maybe today. If you are not in a rush, wait.

blindpet, 2015-03-27:
Kernel clean was already an option and was set to yes by default.
I finally got the PMP working thanks to you!
I don't see the wifi module though, did I need to enable it in the configuration somewhere? modprobe ap6210 says nothing

hvenzke, 2015-03-27:
Igors Images don't support Wifi - bur rasbian do with 3.19 kernels.

root@fw1:/etc/dhcp# lsmod
Module                  Size  Used by
nfsd                  245319  2 
exportfs                3340  1 nfsd
arc4                    1598  2 
rtl8192cu              90112  0 
rtl_usb                11843  1 rtl8192cu
rtl8192c_common        57293  1 rtl8192cu
rtlwifi                81947  3 rtl_usb,rtl8192c_common,rtl8192cu
mac80211              296619  3 rtl_usb,rtlwifi,rtl8192cu
ax88179_178a           13488  0 
usbnet                 23171  1 ax88179_178a
root@fw1:/etc/dhcp#

blindpet, 2015-03-27:
Thanks, I can see I confused the 3.4 kernel with the 3.19. I am trying to build an image that has wifi and port multiplier support. I finally got the PMP support with 3.19 but now no wifi. My next bet is using 4.0 rc I guess, fingers crossed.

igorpecovnik, 2015-03-27:
@Horst
That's not true.

Cubietruck wireless is working in 3.19.x and Banana PRO as well. 

The name of the driver/module is not ap6210 but brcmfmac.

closed 2015-03-27 by igorpecovnik

blindpet, 2015-03-27:
I did modprobe brcmfmac and edited /etc/network/interfaces

lsmod shows the brcmfmac is active but iwconfig does not show any adapters.

igorpecovnik, 2015-03-28:
One more thing.

Check if you are loading proper DTB file. There are banana.dtb and bananapro.dtb ... simply copy "pro" file over "normal" banana and reboot.

blindpet, 2015-03-28:
dtb file in /boot/dtb worked, thanks!