armbian/build #36: Unable to Mount in OpenMediaVault

blindpet, 2015-03-27:
I am using 3.19.3 with the port multiplier enabled and want to use this distro to install omv.

It installs fine but I can't mount any drives, the same drives mount fine on Bananian, the list of errors is here http://forums.openmediavault.org/index.php/Thread/8099-Mount-Error/?postID=74763#post74763 can you point me in the right direction?

igorpecovnik, 2015-03-28:
Probably it's a problem with NFS. I added support into 3.19.x

You can try with this kernel or rebuild your own.
http://mirror.igorpecovnik.com/kernel/3.19.3-cubietruck-next.tar

blindpet, 2015-03-28:
Installed that kernel and same result with mounting errors, any other ideas? I believe OMV supports NFS so I'm not sure it's that.

blindpet, 2015-03-28:
The OMV people have said maybe the kernel has disabled ACL, is that possible or can i recompile the kernel to include it?

blindpet, 2015-03-29:
Any ideas or should I just give up for now? I've been back and forth with the OMV people and they are adamant it's an ACL issue with the kernel.

igorpecovnik, 2015-03-30:
I am not familiar with OpenMediaVault special requirements :( If I knew what to add I would add that already :) You will probably need to experiment - establish build environment and adding things step by step - adding all usually doesn't work.

hvenzke, 2015-04-02:
answering  this for igor:

... from my arm6hf kenrels

root@lb1:~# uname -a
Linux lb1 3.18.7+ #755 PREEMPT Thu Feb 12 17:14:31 GMT 2015 armv6l GNU/Linux

<pre>
root@lb1:~# zcat /proc/config.gz | egrep '(POSIX|ACL|NFS|SMB|ATTR|HFS|EXT|XFS|NTFS|DOS)' | grep -v "#"
CONFIG_BUILDTIME_EXTABLE_SORT=y
CONFIG_POSIX_MQUEUE=y
CONFIG_POSIX_MQUEUE_SYSCTL=y
CONFIG_HAVE_DMA_ATTRS=y
CONFIG_HAVE_CONTEXT_TRACKING=y
CONFIG_MSDOS_PARTITION=y
CONFIG_PAGEFLAGS_EXTENDED=y
CONFIG_ZBOOT_ROM_TEXT=0x0
CONFIG_NET_SCH_HFSC=m
CONFIG_NET_EMATCH_TEXT=m
CONFIG_WIRELESS_EXT=y
CONFIG_WEXT_CORE=y
CONFIG_WEXT_PROC=y
CONFIG_WEXT_SPY=y
CONFIG_WEXT_PRIV=y
CONFIG_CFG80211_WEXT=y
CONFIG_EXTRA_FIRMWARE=""
CONFIG_OF_FLATTREE=y
CONFIG_OF_EARLY_FLATTREE=y
CONFIG_SCSI_ISCSI_ATTRS=y
CONFIG_EXTCON=m
CONFIG_EXTCON_ARIZONA=m
CONFIG_EXT4_FS=y
CONFIG_EXT4_USE_FOR_EXT23=y
CONFIG_EXT4_FS_POSIX_ACL=y
CONFIG_EXT4_FS_SECURITY=y
CONFIG_REISERFS_FS_XATTR=y
CONFIG_REISERFS_FS_POSIX_ACL=y
CONFIG_JFS_POSIX_ACL=y
CONFIG_XFS_FS=m
CONFIG_XFS_QUOTA=y
CONFIG_XFS_POSIX_ACL=y
CONFIG_XFS_RT=y
CONFIG_BTRFS_FS_POSIX_ACL=y
CONFIG_FS_POSIX_ACL=y
CONFIG_MSDOS_FS=y
CONFIG_NTFS_FS=m
CONFIG_NTFS_RW=y
CONFIG_KERNFS=y
CONFIG_TMPFS_POSIX_ACL=y
CONFIG_TMPFS_XATTR=y
CONFIG_HFS_FS=m
CONFIG_HFSPLUS_FS=m
CONFIG_SQUASHFS=m
CONFIG_SQUASHFS_FILE_CACHE=y
CONFIG_SQUASHFS_DECOMP_SINGLE=y
CONFIG_SQUASHFS_XATTR=y
CONFIG_SQUASHFS_ZLIB=y
CONFIG_SQUASHFS_LZO=y
CONFIG_SQUASHFS_XZ=y
HFSPLUS_FS_POSIX_ACL=y
CONFIG_SQUASHFS_FRAGMENT_CACHE_SIZE=3
CONFIG_F2FS_FS_XATTR=y
CONFIG_F2FS_FS_POSIX_ACL=y
CONFIG_NFS_FS=y
CONFIG_NFS_V2=y
CONFIG_NFS_V3=y
CONFIG_NFS_V3_ACL=y
CONFIG_NFS_V4=y
CONFIG_NFS_SWAP=y
CONFIG_ROOT_NFS=y
CONFIG_NFS_FSCACHE=y
CONFIG_NFS_USE_KERNEL_DNS=y
CONFIG_NFSD=m
CONFIG_NFSD_V2_ACL=y
CONFIG_NFSD_V3=y
CONFIG_NFSD_V3_ACL=y
CONFIG_NFSD_V4=y
CONFIG_NFS_ACL_SUPPORT=y
CONFIG_NFS_COMMON=y
CONFIG_CIFS_XATTR=y
CONFIG_CIFS_POSIX=y
CONFIG_9P_FS_POSIX_ACL=y
CONFIG_CONTEXT_SWITCH_TRACER=y
CONFIG_TEXTSEARCH=y
CONFIG_TEXTSEARCH_KMP=m
CONFIG_TEXTSEARCH_BM=m
CONFIG_TEXTSEARCH_FSM=m
CONFIG_NLATTR=y

</pre>

igorpecovnik, 2015-05-02:
It should work now.

blindpet, 2015-05-02:
With 3.19.3 and your ready made image or I have to build my own as hvenze outlined?

igorpecovnik, 2015-05-02:
Oh, I mixed this issue with another one. Not merged yet :( but you can just copy / past this to the config and try to compile. We are now on 3.19.6 ...

igorpecovnik, 2015-05-02:
Current configuration is now updated. I also compiled new kernel for Bananapi with all this.
http://mirror.igorpecovnik.com/kernel/3.19.6-bananapi-next.tar

RafaelKa, 2015-06-07:
try to remove all quota options from `/etc/fstab` manually for your partition and then run
```mount -a```

hvenzke, 2015-06-20:
my current R1 3.19.3 kernel for ext 
<pre>
rnot@fw1 ~ $ zcat  /proc/config.gz | egrep '(ACL|XAT|EXT2|EXT3|EXT4)' 
# CONFIG_EXT2_FS is not set
# CONFIG_EXT3_FS is not set
CONFIG_EXT4_FS=y
CONFIG_EXT4_USE_FOR_EXT23=y
CONFIG_EXT4_FS_POSIX_ACL=y
CONFIG_EXT4_FS_SECURITY=y
# CONFIG_EXT4_DEBUG is not set
CONFIG_FS_POSIX_ACL=y
CONFIG_TMPFS_POSIX_ACL=y
CONFIG_TMPFS_XATTR=y
# CONFIG_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_SQUASHFS_XATTR is not set
CONFIG_NFS_V3_ACL=y
CONFIG_NFSD_V2_ACL=y
CONFIG_NFSD_V3_ACL=y
CONFIG_NFS_ACL_SUPPORT=y
# CONFIG_CIFS_XATTR is not set
</pre>

@RafaelKa please check your kernel options compiled in.

blindpet, 2015-06-20:
Seems to be working on 3.19.7

hvenzke, 2015-06-20:
@blindpet  thanks for report . will see hows going with 4.x mainlain .. on the way updating just

closed 2016-03-11 by igorpecovnik