armbian/build #39: make-kpkg  FAIL  on debian  8 :

hvenzke, 2015-04-02:
Dear igor .

please fix the debian pkg build for kernel 3.19.2 and up.

fakeroot make-kpkg buildpackage --initrd --revision 04 &
caused to  fail at  3.19.3 from your "lib patched " sources .

this seems to comes from include $(DEBDIR)/ruleset/targets/doc.mk at debian/ruleset/targets/common.mk.

assuming that an patch exist allready somewhere ..

<pre>

...
(cd /usr/src/linux/debian/linux-source-3.19.3bpi-dirty/usr/src/linux-source-3.19.3bpi-dirty; find . -type d -name .arch-ids -print0  | xargs -0r rm -rf {} \; )
(cd /usr/src/linux/debian/linux-source-3.19.3bpi-dirty/usr/src/linux-source-3.19.3bpi-dirty; find . -type d -name .git -print0       | xargs -0r rm -rf {} \; )
(cd /usr/src/linux/debian/linux-source-3.19.3bpi-dirty/usr/src/linux-source-3.19.3bpi-dirty; find . -type f -name .gitmodule -print0 | xargs -0r rm -f  {} \; )
(cd /usr/src/linux/debian/linux-source-3.19.3bpi-dirty/usr/src/linux-source-3.19.3bpi-dirty && cd .. &&                                            \
           tar --bzip2 -cf linux-source-3.19.3bpi-dirty.tar.bz2 linux-source-3.19.3bpi-dirty && \
             rm -rf linux-source-3.19.3bpi-dirty;)
make[2]: Leaving directory '/usr/src/linux'
debian/ruleset/common/targets.mk:357: recipe for target 'debian/stamp/do-install-indep' failed
make[1]: *** [debian/stamp/do-install-indep] Error 2
make[1]: Leaving directory '/usr/src/linux'
dpkg-buildpackage: Fehler: Fehler-Exitstatus von debian/rules binary war 2
debian/ruleset/targets/common.mk:401: recipe for target 'debian/stamp/build/buildpackage' failed
make: *** [debian/stamp/build/buildpackage] Error 2
</pre>

igorpecovnik, 2015-04-02:
I just compiled 3.19.3 kernel and packaging works well. Anyway this is the only relevant patch:
https://github.com/igorpecovnik/lib/blob/next/patch/packaging-next.patch

hvenzke, 2015-04-02:
Ok i'll fetch new sources  and try again.

The only part what im using are your patched kernel source base as there are currently no other then you seems work on the NEXT BPi kernels currently.

closed 2015-05-02 by igorpecovnik