armbian/build #41: Can4linux Compile but Crash with Fatal error

kaneghost, 2015-04-02:
root@bananapi:~/can4linux-code/can4linux# insmod can4linux.ko

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1282.660992] Internal error: Oops: 5 [#1] PREEMPT SMP ARM

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1283.974756] Process insmod (pid: 2601, stack limit = 0xee6ee2f0)

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1283.986480] Stack: (0xee6efed8 to 0xee6f0000)

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.001982] fec0:                                                       eeab5a40 bf0b6da4

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.017782] fee0: 00000000 ee67b380 00000001 c0083eb0 bf0b6db0 00007fff 00000000 ee6effb0

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.033712] ff00: b6f52d50 00000000 00000000 c0081bd8 ee6ee030 f02317a0 b6f52d50 f02300f4

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.049713] ff20: f0231250 f0231778 eea01900 00000000 00000000 00000000 00000000 00000000

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.065816] ff40: 00000000 f022b000 0000a168 f0231250 f02310f3 f02346b0 0000f598 00010538
Segmentation fault

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.081805] ff60: 00000000 00000000 00000021 00000022 00000016 0000001a 0000000f 00000000

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.097951] ff80: 08000071 00000000 b8d22df8 b8d22d40 00000080 c000eac4 ee6ee000 00000000
root@bananapi:~/can4linux-code/can4linux# 
Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.113971] ffa0: 00000000 c000e940 00000000 b8d22df8 b6e32000 0000a168 b6f52d50 00000002

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.129915] ffc0: 00000000 b8d22df8 b8d22d40 00000080 00000000 b6f52d50 0000a168 00000000

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.145809] ffe0: 00000000 bec4fafc b6f4cb07 b6ed0264 80000010 b6e32000 00000000 00000000

Message from syslogd@localhost at Apr  2 12:26:40 ...
 kernel:[ 1284.192789] Code: e595315c e5911014 e2833004 e7922101 (e7931002)

hvenzke, 2015-04-10:
Driver sources : http://www.port.de/de/produkte/canopen/canopen-software/linux-can-driver.html
testing load module failed - depend on missing CAN support .
recompile 3.19.3 with CAN support under way.

hvenzke, 2015-04-10:
<pre>
root@fw1 /var/log # find /lib/modules/3.19.3-bpi-dirty/ | grep can
/lib/modules/3.19.3-bpi-dirty/kernel/net/can
/lib/modules/3.19.3-bpi-dirty/kernel/net/can/can-gw.ko
/lib/modules/3.19.3-bpi-dirty/kernel/net/can/can-bcm.ko
/lib/modules/3.19.3-bpi-dirty/kernel/net/can/can.ko
/lib/modules/3.19.3-bpi-dirty/kernel/net/can/can-raw.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/flexcan.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/softing
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/softing/softing.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/rcar_can.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/grcan.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/m_can
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/m_can/m_can.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/ti_hecc.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/c_can
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/c_can/c_can.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/vcan.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/spi
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/spi/mcp251x.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/can-dev.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/slcan.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/usb
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/usb/usb_8dev.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/usb/peak_usb
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/usb/peak_usb/peak_usb.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/usb/ems_usb.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/usb/esd_usb2.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/usb/kvaser_usb.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/usb/gs_usb.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/cc770
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/cc770/cc770.ko
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/sja1000
/lib/modules/3.19.3-bpi-dirty/kernel/drivers/net/can/sja1000/sja1000.ko

~:  /var/log # modprobe can
:~  /var/log # modprobe can-gw
~: /var/log # modprobe can-raw

~: /var/log # lsmod | grep can
can_raw                 5728  0 
can_gw                  7280  0 
can                    25743  2 can_gw,can_raw
 ~: /var/log # uname -a
Linux fw1 3.19.3-bpi-dirty #1 SMP Fri Apr 10 18:28:29 CEST 2015 armv7l GNU/Linux

[  300.610294] can: controller area network core (rev 20120528 abi 9)
[  300.610428] NET: Registered protocol family 29
[  422.722429] can: netlink gateway (rev 20130117) max_hops=1
[  429.915780] can: raw protocol (rev 20120528)

 ~ # 
</pre>

issue solved - 3.19.3 supports CAN bus out of the box - if compiled in

igorpecovnik, 2015-09-11:
otherwise can4linux is able to compile on 3.4.108 / bananapi

closed 2015-09-11 by igorpecovnik