armbian/build #42: sunxi-tools.  issue with lib and 3.19.9 kernel

hvenzke, 2015-04-02:
pushed compile tests fwd to kernel 3.19.9 resulted in 

------ Downloading sunxi-tools.
Already up-to-date.
error: pathspec 'v3.19.9' did not match any file(s) known to git.
patching file scripts/Makefile.dtbinst
patching file scripts/package/builddeb
Hunk #6 succeeded at 362 (offset 11 lines).
patching file arch/arm/boot/dts/Makefile
Hunk #1 FAILED at 464.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm/boot/dts/Makefile.rej
patching file arch/arm/boot/dts/sun7i-a20-bananapipro.dts
patching file arch/arm/boot/dts/Makefile
Hunk #1 FAILED at 465.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm/boot/dts/Makefile.rej
patching file arch/arm/boot/dts/sun7i-a20-lamobo-r1.dts
patching file arch/arm/boot/dts/x-powers-axp209.dtsi
patching file drivers/net/phy/Kconfig

igorpecovnik, 2015-04-02:
error: pathspec 'v3.19.9' did not match any file(s) known to git.

What's the problem here?

hvenzke, 2015-04-02:
use this:

# advanced
BRANCH="next"
KERNELTAG="v3.19.9"
FBTFT="yes"                                 # https://github.com/notro/fbtft
EXTERNAL="yes"                              # compile extra drivers`

then try yourself

igorpecovnik, 2015-04-02:
I think even with those errors it should complete compilation but without R1 DTB file ;) 

Anyway. I should note that choosing not existing tag means staying on previously selected or default and this can result in error. Or if I put this into another words - leave default if you are not sure.

Patch for adding R1 is not compatible to source 4.x ... but I made a patch which at this point need to be applied manually. It's under /patch

closed 2015-04-02 by igorpecovnik

hvenzke, 2015-04-02:
as debian8 on GOLIVE very soon , thats why i am trying prepated for the 4.x