armbian/build #45: 3.19.3 depmod

hvenzke, 2015-04-09:
...

make modules_install INSTALL_MOD_PATH=/
<pre>
...
 DEPMOD  3.19.3-bpi-dirty
depmod: ERROR: Found 5 modules in dependency cycles!
depmod: ERROR: Cycle detected: drm
depmod: ERROR: Cycle detected: sti_vtg -> drm
*** Error in `/sbin/depmod': malloc(): memory corruption: 0xb712d170 ***
./scripts/depmod.sh: Zeile 57:  8163 Abgebrochen             "$DEPMOD" "$@" "$KERNELRELEASE" $SYMBOL_PREFIX
Makefile:1138: recipe for target '_modinst_post' failed
make: *** [_modinst_post] Error 134
~ # 
</pre>

hvenzke, 2015-04-09:
well , Igor - 
please take the latest code from http://code.metager.de/source/xref/linux/stable/drivers/gpu/drm/sti/

igorpecovnik, 2015-04-09:
OK, BTW - do you have any better / fixed DTB for R1? I made some test build & boot with the one from openwrt but it looks that definitions are not correct - no USB, no SATA power, ...

hvenzke, 2015-04-09:
answer for your question is offtopic this issue .. lets move this to seperate issue.

hvenzke, 2015-04-10:
solved temporaly with disabling compiling the  sti_vtg till new driver pullled by igor.

closed 2016-06-06 by zador-blood-stained