armbian/build #46: 3.19.3 DTB problem Analyze

hvenzke, 2015-04-09:
Igor asked : do you have any better / fixed DTB for R1? I made some test build & boot with the one from openwrt but it looks that definitions are not correct - no USB, no SATA power, ..."

hvenzke, 2015-04-09:
Igor - i am on automated kernel Build & install tests with the 3.19.3 at current.

<pre>
root@fw1 /boot # ls -la
insgesamt 26804
drwxr-xr-x  4 root root    16384 Apr  9 11:40 .
drwxr-xr-x 27 root root     4096 Apr  9 11:34 ..
-rwxr-xr-x  1 root root    17856 Mär 26 17:52 bootcode.bin
-rwxr-xr-x  1 root root      298 Feb  8 18:05 boot.scr
-rwxr-xr-x  1 root root       97 Feb  8 10:44 cmdline.txt
-rwxr-xr-x  1 root root   124236 Apr  9 11:35 config-3.19.3-bpi-dirty
-rwxr-xr-x  1 root root     1322 Feb  8 10:45 config.txt
-rwxr-xr-x  1 root root    18693 Mär 26 17:52 COPYING.linux
drwxr-xr-x  2 root root     8192 Apr  9 11:40 dtb
-rwxr-xr-x  1 root root     2345 Mär 26 17:52 fixup_cd.dat
-rwxr-xr-x  1 root root     6135 Mär 26 17:52 fixup.dat
-rwxr-xr-x  1 root root     9191 Mär 26 17:52 fixup_x.dat
-rwxr-xr-x  1 root root      137 Jan 31 20:24 issue.txt
-rwxr-xr-x  1 root root  3969080 Mär 26 17:52 kernel7.img
-rwxr-xr-x  1 root root 11006448 Apr  9 11:35 kernel.img
-rwxr-xr-x  1 root root     1447 Mär 26 17:52 LICENCE.broadcom
-rwxr-xr-x  1 root root    18974 Sep 25  2013 LICENSE.oracle
drwxr-xr-x  2 root root     8192 Mär 26 17:52 overlays
-rwxr-xr-x  1 root root   554520 Mär 26 17:52 start_cd.elf
-rwxr-xr-x  1 root root  2642392 Mär 26 17:52 start.elf
-rwxr-xr-x  1 root root  3599400 Mär 26 17:52 start_x.elf
-rwxr-xr-x  1 root root    23406 Apr  9 11:35 sun7i-a20-bananapi.dtb
-rwxr-xr-x  1 root root  5311656 Apr  9 11:35 uImage
</pre>

hvenzke, 2015-04-10:
kernel 3.19.3 are stable running from last nigtly <09-04-2015> build :
<pre>
rnot@fw1 ~ $ uname -a
Linux fw1 3.19.3-bpi-dirty #1 SMP Fri Apr 10 02:29:57 CEST 2015 armv7l GNU/Linux
rnot@fw1 ~ $ lsmod
Module                  Size  Used by
nfsd                  235118  2 
exportfs                3257  1 nfsd
arc4                    1494  2 
rtl8192cu              89283  0 
rtl_usb                11672  1 rtl8192cu
rtl8192c_common        55660  1 rtl8192cu
rtlwifi                79786  3 rtl_usb,rtl8192c_common,rtl8192cu
mac80211              287615  3 rtl_usb,rtlwifi,rtl8192cu
ax88179_178a           13266  0 
leds_gpio               2848  0 
usbnet                 22637  1 ax88179_178a
~ $ 
</pre>

DTB file

hvenzke, 2015-04-10:
attached 120G ssd  to the R1 

<pre>
dmesg 
..
[10813.099436] ata1: exception Emask 0x10 SAct 0x0 SErr 0x4050002 action 0xe frozen
[10813.099555] ata1: irq_stat 0x00000040, connection status changed
[10813.099568] ata1: SError: { RecovComm PHYRdyChg CommWake DevExch }
[10813.099606] ata1: hard resetting link
[10814.923946] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[10814.924700] ata1.00: ATA-9: OCZ-VERTEX4, 1.5, max UDMA/133
[10814.924723] ata1.00: 250069680 sectors, multi 16: LBA48 NCQ (depth 31/32)
[10814.925486] ata1.00: configured for UDMA/133
[10814.925542] ata1: EH complete
[10814.925930] scsi 0:0:0:0: Direct-Access     ATA      OCZ-VERTEX4      1.5  PQ: 0 ANSI: 5
[10814.928109] sd 0:0:0:0: [sda] 250069680 512-byte logical blocks: (128 GB/119 GiB)
[10814.928293] sd 0:0:0:0: [sda] Write Protect is off
[10814.928309] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[10814.928370] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[10814.935163]  sda:
[10814.936227] sd 0:0:0:0: [sda] Attached SCSI disk
[10814.995912] sd 0:0:0:0: Attached scsi generic sg0 type 0
[10874.359582]  sda: sda1

~: /t/GIT/output/linux-mainline/include # lsmod
Module                  Size  Used by
sg                     28062  0 
nfsd                  235118  2 
exportfs                3257  1 nfsd
arc4                    1494  2 
rtl8192cu              89283  0 
rtl_usb                11672  1 rtl8192cu
rtl8192c_common        55660  1 rtl8192cu
rtlwifi                79786  3 rtl_usb,rtl8192c_common,rtl8192cu
mac80211              287615  3 rtl_usb,rtlwifi,rtl8192cu
ax88179_178a           13266  0 
leds_gpio               2848  0 
usbnet                 22637  1 ax88179_178a

root@fw1 /t/GIT/output/linux-mainline/include # iostat -x
Linux 3.19.3-bpi-dirty (fw1) 	10.04.2015 	_armv7l_	(2 CPU)

avg-cpu:  %user   %nice %system %iowait  %steal   %idle
          24,43    0,00    5,20    1,17    0,00   69,21

Device:         rrqm/s   wrqm/s     r/s     w/s    rkB/s    wkB/s avgrq-sz avgqu-sz   await r_await w_await  svctm  %util
mmcblk0           0,38     0,50    0,41    0,40    10,10     6,01    39,95     0,01   16,87    2,84   31,06   4,43   0,36
dm-0              0,00     0,00    0,01    0,00     0,04     0,00     8,80     0,00    6,02    6,00    6,67   0,75   0,00
sda               0,00   100,05    0,02    0,61     0,10   400,88  1264,00     0,40  628,55    0,04  652,46  21,06   1,34

oot@fw1 /t/GIT/output/linux-mainline/include # 10240+0 Datensätze ein
10240+0 Datensätze aus
5368709120 Bytes (5,4 GB) kopiert, 177,345 s, 30,3 MB/s

real	2m57.392s
user	0m0.110s
sys	0m34.810s
^C
[1]+  Fertig                  time dd if=/dev/zero of=/dev/sda1 bs=512k count=10240

</pre>

hvenzke, 2015-04-10:
usb storage works fine with 3.19.3 - testbuild3 today.

<pre>
# lsmod | grep usb
usb_storage            50808  1 uas
usbnet                 22661  1 ax88179_178a
rtl_usb                11776  1 rtl8192cu
rtlwifi                80106  3 rtl_usb,rtl8192c_common,rtl8192cu
mac80211              287759  3 rtl_usb,rtlwifi,rtl8192cu
</pre>

igorpecovnik, 2015-05-02:
Fixed DTB is a part of latest image now. Closing.

closed 2015-05-02 by igorpecovnik