armbian/build #52: SPI patches

JHSawatzki, 2015-04-27:
What is the difference between spi.patch and spi-sun7i.patch? Some time ago you were using spi-sun7i.patch and then spi.patch. I think spi.patch has some typos on lines 574 and 576.
Where do you got the spi.patch from? I can only verify the changes applied in spi-sun7i.patch from other common sunxi sources.

unsigned char spi_tx_ndrq[] = {N_DST_SPI0_TX, N_DST_SPI1_TX, **N_DST_SPI3_TX**, N_DST_SPI3_TX};

unsigned char spi_tx_ddrq[] = {D_DST_SPI0_TX, D_DST_SPI1_TX, **D_DST_SPI3_TX**, D_DST_SPI3_TX};

Thanks for your hard work!

JHSawatzki, 2015-05-02:
Seems to be a typo.
ref: https://github.com/OLIMEX/OLINUXINO/commit/719003c3f6964b9373198bf67e9e90294848e10e

zador-blood-stained, 2016-06-04:
Current SPI patches don't have this issue.

closed 2016-06-04 by zador-blood-stained