armbian/build #53: fix video memory for fbturbo x driver

Trinity93, 2015-05-14:
20.936] (II) FBTURBO(0): tear-free zero-copy double buffering needs more video memory
[    20.936] (II) FBTURBO(0): please set fb0_framebuffer_num >= 3 in the fex file
[    20.936] (II) FBTURBO(0): and sunxi_fb_mem_reserve >= 24 in the kernel cmdline

 trying to figure out how to fix this in the lambo r1 weasy image

igorpecovnik, 2015-05-15:
Edit /boot/boot.cmd and alter numbers. Remember to recompile to boot.scr.

Trinity93, 2015-05-15:
ty for getting back to me. i figured it out and got fbturbo driver working but  in the end it didn’t realy help video performance much.  wants something called lime to do it which seems only available for android

closed 2015-05-15 by igorpecovnik