armbian/build #55: Various issues and resolves

db260179, 2015-05-16:
Hi Igor,

Thanks for the good work you are doing. You are my main source on what is going on with these boards these days.

Just to help you out, the following issues have been found:

1. tusbd doesnt compile for linux kernel 4.0 anymore - src/tusbd/minor_control.c:54:1: error: implicit declaration of function ‘strnicmp’ [-Werror=implicit-function-declaration]
Alternative is to use the usbip - can compile module in the kernel. http://usbip.sourceforge.net/

2.Please see attached patch to enable the axp20x power button so you can poweroff the boards.
Just rename to .patch

You also need to change the acpi function on Debian Jessie as current sh doesnt work correctly.

cat /etc/acpi/events/powerbtn-acpi-support 
event=button/power
#action=/etc/acpi/powerbtn-acpi-support.sh
action=/etc/acpi/powerbtn.sh %e

cat /etc/acpi/powerbtn.sh 
#!/bin/sh
/sbin/shutdown -h -P now "Power button pressed"
exit 0

[image]

Thanks

igorpecovnik, 2015-05-16:
Thanks! 

1. I noticed troubles about tusbd but am to busy to look into. Changing to usbip also looks problematic since it hasn't been maintained for a while.

2. Can't download even if I rename it or I miss something. Huh. Maybe just add a patch directly to sources?

db260179, 2015-05-19:
Ok, Added the pull request for the patch
https://github.com/igorpecovnik/lib/pull/57

igorpecovnik, 2015-05-20:
This driver is already present in 4.0.4 ;)

closed 2015-05-26 by db260179