armbian/build #58: cubietruck (Debian Wheezy, Jessie or Ubuntu Trusty based) Kernel 4.0.3, 14,05,2015

DrakonX, 2015-05-20:
Good day, I tried to use for cubietruck (Debian Wheezy, Jessie or Ubuntu Trusty based) Kernel 4.0.3, and each time he hung (or stopped and not continued) on the line [ok] restarting openbsd secure shell server sshd.

igorpecovnik, 2015-05-20:
Yes, I noticed. There are some troubles with Ethernet. Sometimes is not deteced. Power on / off helps ... use one version back and you will be fine.

DrakonX, 2015-05-20:
thank you very much, I just fell back to the version before, in general thank you for the great assembly.

closed 2015-05-20 by DrakonX

cibomato, 2015-06-09:
Hi,

same here with Debian wheezy, 4.0.4. eth0 doesn't come up and get stuck at restarting sshd. Powercycle doesn't help. 
In latest changelog, I've read that ethernet init was fixed, could there still be a problem?
Or do you suggest to use another image?

Thanks for your great work and best regards,
Jochen