armbian/build #62: Ubuntu Image, 4.0.4 Kernel compiled with USB_STORAGE_DEBUG

SgtSeppel, 2015-06-08:
I just installed the above mentioned image from http://mirror.igorpecovnik.com/Cubietruck_Ubuntu_3.8_trusty_4.0.4.zip.

Installing my HUAWEI USB modem is virtually impossible due to the fact that syslog is spammed with cyclic ouput of the USB storage subsystem (the modem is recognized as a storage device).

zcat /config.gz | grep "USB_STORAGE_DEBUG" reveals that debugging is enabled in the kernel.

Is that intended?

igorpecovnik, 2015-06-08:
No, by mistake. It should already be out of the default config.

SgtSeppel, 2015-06-08:
thanks for the quick reply. So this will be fixed in the next release?
Or maybe I'll try to patch together a working build environment...

Anyway, keep up the good work!

igorpecovnik, 2015-06-08:
Thanks. Build environment is already fixed, download and compile your own kernel or you can use this already fixed kernel: http://mirror.igorpecovnik.com/kernel/4.0.4-orangepi-next.tar
It's for Orange but it's the same. Then only difference is name.

closed 2015-06-08 by igorpecovnik