armbian/build #63: swconfig

fine77, 2015-06-09:
i would suggest to integrate the swconfig source into the build process.
i had some issues with the integrated swconfig.tgz.

i've used the source code from https://github.com/jekader/swconfig with success.

hvenzke, 2015-06-20:
i recomend that too from today rebuild  the 4.x kernel only mainline.

@igorpecovnik  please update scripting to dynamic build the swconfig when kernel been build .

regards
Horst

igorpecovnik, 2015-06-20:
I already try but couldn't CC, only native. So the task is stalled / not resolved yet.

hvenzke, 2015-06-20:
lets think how to do this .. this wuold just one more sub-section for R1 
   ... as your lib does it equal way for many other thinks as ..
  -> linux-mainline  ,  rt8192cu ,  sunxi-display-changer ,  sunxi-tools ,  u-boot

- git clone the https://github.com/jekader/swconfig into output/swconfig
- add patches 
- compile 

i'll give it an try once my 4.x next kernel are ready .

igorpecovnik, 2015-06-21:
Yes, that's the idea. CC must work ;)

zador-blood-stained, 2016-06-20:
Implemented via building in chroot: https://github.com/igorpecovnik/lib/commit/1fd33d373e82de7505fa476b6912a49d82db2a6c

Packages will be added to repository eventually.

closed 2016-06-20 by zador-blood-stained