armbian/build #64: build errors with udoo quad/next and current version 186ccb

spinxz, 2015-06-10:
I get build errors when trying to create a 4.0 kernel for the udoo quad (debian jessie).
I am using your compile.sh script adapted as follows:

```
KERNEL_ONLY="yes" 
SOURCE_COMPILE="yes" 
KERNEL_CONFIGURE="yes" 
KERNEL_CLEAN="yes"
USEALLCORES="yes" 
BUILD_DESKTOP="yes"

DEST_LANG="en_US.UTF-8"                     # sl_SI.UTF-8, en_US.UTF-8
CONSOLE_CHAR="UTF-8"                        # console charset
TZDATA="Europe/Berlin" 
```

At the beginning I get:
```
 [ok ] Downloading linux-udoo-next
error: pathspec 'master' did not match any file(s) known to git.
```

Then at the end of the build process:
```
  ...
  Kernel: arch/arm/boot/zImage is ready
  DTC     arch/arm/boot/dts/imx6dl-aristainetos_4.dtb
Error: arch/arm/boot/dts/imx6dl-aristainetos_4.dts:85.2-3 label or path, 'ipu1_di0_disp0', not found
FATAL ERROR: Syntax error parsing input tree
make[3]: *** [arch/arm/boot/dts/imx6dl-aristainetos_4.dtb] Fehler 1
make[2]: *** [dtbs] Fehler 2
make[1]: *** [deb-pkg] Fehler 2
make: *** [deb-pkg] Fehler 2
[ ok ] Kernel building done
[ ok ] Target directory: /home/tom/udoo/output/output/kernel
[ ok ] File name: 4.0.0-udoo-next.tar
```

In the output folder only the *linux-u-uboot-4.0.0-udoo_1.7_armhf.deb* is built. The kernel is missing.

Apart from that - thanks so much for your great work! The udoo was useless to me until I found your images...

igorpecovnik, 2015-06-11:
Building is failing because patch is not designed to work with kernel other than 3.18 ... there are many ways how to fix this. If you look into patch:
https://github.com/igorpecovnik/lib/blob/next/patch/udoo-dts-fix.patch
Patch is removing all dtbs except udoo. Other way is to fix / remove only the failed one(s): imx6dl-aristainetos_4.dts or to compile only dtbs of your wish. In all cases - manual fixing + creating new patch or fixing the script.

spinxz, 2015-06-12:
Thanks for your answer. I adapted the section *dtb-$(CONFIG_SOC_IMX6Q)* in the Makefile in *linux-udoo-next/arch/arm/boot/dts/* to:
```
dtb-$(CONFIG_SOC_IMX6Q) += \
	imx6dl-udoo.dtb \
	imx6q-udoo.dtb \
```
So everything unrelated to udoo got removed. However, I failed to find/fix the proper make rules and compiling reports

    make[3]: *** No rule available to build the target »arch/arm/boot/dts/dtb-y«, 
        required by »__build« 

Any idea what else needs to be changed?

igorpecovnik, 2015-06-12:
targets += dtbs dtbs_install
    targets += $(dtb-y)

Add this.

closed 2015-06-12 by igorpecovnik

spinxz, 2015-06-12:
Its working now. I had the rule but made another typo (forgot to remove a `\`).
Thanks a lot.