armbian/build #65: no CPU temperature on A20 Kernel 4.0.4

holsteiner, 2015-06-11:
Banana PI with kernel 4.0.4 cannot show CPU temperature

igorpecovnik, 2015-06-11:
Kernel 4.x doesn't have functionality of 3.4.x yet. Check here:
http://linux-sunxi.org/Linux_mainlining_effort

I am not aware that CPU temperature is already done.

holsteiner, 2015-06-11:
Sorry, for asking again...
I2C had been merged in to main line  kernel 3.13...
http://linux-sunxi.org/Linux_mainlining_effort#Merged_into_3.13
Why it had been removed?

igorpecovnik, 2015-06-11:
I2C is present and operational within kernel but I wasn't testing with Banana but Orange and Lime.

holsteiner, 2015-06-11:
sorry again, I wasn't precise enough:
I thought temperature measurement is based on I2C...
in kernel 3.4.x it can be read this way:
cat /sys/devices/platform/sunxi-i2c.0/i2c-0/0-0034/temp1_input

sorry, I don't know much about Linux kernel stuff... I just try to understand...
...it seems to be a special sunxi implementation...!?

igorpecovnik, 2015-06-11:
Ahaa, true .. In this case it should be possible.

holsteiner, 2015-06-11:
A list of installed busses:

Banana PI Kernel 4.0.4
i2cdetect -l
i2c-0   i2c             mv64xxx_i2c adapter                     I2C adapter
i2c-1   i2c             mv64xxx_i2c adapter                     I2C adapter

Banana PI Kernel 3.4.90
i2cdetect -l
i2c-0   i2c             sunxi-i2c.0                             I2C adapter
i2c-1   i2c             sunxi-i2c.1                             I2C adapter
i2c-2   i2c             sunxi-i2c.2                             I2C adapter
i2c-3   i2c             sunxi-i2c.3                             I2C adapter
i2c-4   i2c             sunxi-hdmi-i2c                       I2C adapter

holsteiner, 2015-06-12:
is CONFIG_CPU_THERMAL=y set in config for BananaPI?
In linux-sunxi-next.config and linux-sunxi.config it is not set...
I cannot check it now, I am at work...

igorpecovnik, 2015-06-12:
I am not sure if this wayis done yet ... but there must be like this script which is getting info out of AXP chip:

http://learn.linksprite.com/pcduino/pcduino-hardware-basic-and-improve/axp209-battery-info-with-a-script/

holsteiner, 2015-06-12:
just fyi:
sunxi-next kernel 3.19.rc... has:
CONFIG_THERMAL=y 
81 CONFIG_CPU_THERMAL=y 
https://github.com/linux-sunxi/linux-sunxi/blob/sunxi-next/arch/arm/configs/sunxi_defconfig

holsteiner, 2015-06-12:
I have found this, but I don't if it is still relevant:
A20 SoC TP sensor controller: add drivers/hwmon/a20-tp.c
https://groups.google.com/forum/message/raw?msg=linux-sunxi/2zoxTcepPVo/q4f3D1ZEm8cJ

igorpecovnik, 2015-06-12:
Relevant to kernel 3.4 where we have have both options - from A20 and AXP, both are working in my kernel. For kernel 4.x this is just one of many opened issues - general speaking.

igorpecovnik, 2015-06-16:
https://github.com/igorpecovnik/lib/commit/5ef8f7386eab398e886fe6edf041b0f8754350f3

closed 2015-06-16 by igorpecovnik

holsteiner, 2015-06-16:
thanks a lot... it's working :+1: