armbian/build #67: NEXT : bug with 4.x  xen hypervisor  + powermanagment

hvenzke, 2015-06-20:
dear @igorpecovnik 

please fetch the lastest arm-xen + hypervisor  + powermanagment into your automated NEXT patch series.

<pre>
arch/arm/kernel/built-in.o: In function `cpu_resume':
psci_smp.c:(.data+0x1ca4): relocation truncated to fit: R_ARM_THM_CALL against symbol `__hyp_stub_install_secondary' defined in .text section in arch/arm/kernel/built-in.o
Makefile:933: recipe for target 'vmlinux' failed
make[2]: *** [vmlinux] Error 1
scripts/package/Makefile:90: recipe for target 'deb-pkg' failed
make[1]: *** [deb-pkg] Error 2
Makefile:1238: recipe for target 'deb-pkg' failed
make: *** [deb-pkg] Error 2
</pre>

igorpecovnik, 2015-06-20:
CPU frequency scaling is working out of the box. What else?

hvenzke, 2015-06-20:
ill report more as far as my current build run compile completed, first try aborted with above error

hvenzke, 2015-06-21:
@igorpecovnik  

<pre>
Image Name:   Linux-4.0.5bpi-dirty
Created:      Sun Jun 21 01:28:18 2015
Image Type:   ARM Linux Kernel Image (uncompressed)
Data Size:    5123360 Bytes = 5003.28 kB = 4.89 MB
Load Address: 40008000
Entry Point:  40008000
  Image arch/arm/boot/uImage is ready
Kernel build done
</pre>

kernel boot , but, 
1. seems that the CONFIG_USB_OTG_WHITELIST are buggy with 4.0.5
  currently build under work without that WHITELIST
2. seems that 4.0.5 firmware upload / handling take long time or maybe broken

hvenzke, 2015-06-23:
@igorpecovnik 
well said to say - but will stop my experiements in banana PI R1 as repeatelty the kenrel updates fail again and again .

My need are for an rock-solid solution - and The R1 are today NOT this way.

igorpecovnik, 2015-06-23:
Yes, there are bugs in the kernel but R1 also needs some HW fixing .. they are already working on it: http://wiki.lamobo.org/index.php/Lamobo_R1S

closed 2015-09-11 by igorpecovnik