armbian/build #68: Add options to prevent installation of optional packages, custom MOTD, etc.

n-st, 2015-06-22:
Hi,

could you please add an option (or perhaps a Git branch) that prevents the script from installing this [extensive list of optional software](https://github.com/igorpecovnik/lib/blob/next/deboostrap.sh#L113)?
An option to disable the customized MOTD, the automatic partition resize on first boot, and the hostapd patching in distributions.sh would also be very helpful.

Your script is a great way to effortlessly create a fairly minimal Debian image, but sadly it then goes ahead and applies a bunch of modifications that are likely to be unnecessary or even unwanted for some users…

igorpecovnik, 2015-06-23:
I can move things into this direction but slowly / low priority - just about to leave for summer vacations. You are welcome to assist if you find time. I fully support the idea.

zador-blood-stained, 2016-06-04:
Current build script allows customizing package lists and defining custom actions during build process, so it should be possible to create minimal enough distributions without making modifications to code.

closed 2016-06-04 by zador-blood-stained