armbian/build #74: Unable to make scripts after Kernel update

berturion, 2015-07-11:
Hello Igor,
I used your great script to build a 4.1.1 kernel only. I installed it on an existing jessie os that I took from your blog post with kernel 4.0.5. The new 4.1.1 kernel is now installed with headers... etc but, as I need dvb-usb-dib700 module for my PlayTV tuner, I need to compile the module it and to do so, I need to `make scripts` from the directory `/usr/src/linux-headers-4.1.1-cubieboard2`. Unfortunately the result is this:

     CHECK   usr/include/video/ (3 files)
      CHECK   usr/include/xen/ (4 files)
      CHECK   usr/include/uapi/ (0 files)
      CHECK   usr/include/asm/ (34 files)
      CHK     include/generated/uapi/linux/version.h
      HOSTCC  scripts/sortextable
    scripts/sortextable.c:31:32: fatal error: tools/be_byteshift.h: No such file or directory
     #include <tools/be_byteshift.h>
                                    ^
    compilation terminated.
    scripts/Makefile.host:91: recipe for target 'scripts/sortextable' failed
    make[1]: *** [scripts/sortextable] Error 1
    Makefile:555: recipe for target 'scripts' failed
    make: *** [scripts] Error 2

Could you help me please ?

igorpecovnik, 2015-07-12:
Use this patch on the top of headers tree
https://github.com/igorpecovnik/lib/blob/next/patch/headers-debian-byteshift.patch

patch -p1 headers-debian-byteshift.patch

then make scripts again and it should be fine. I need to add this to the manual asap.

berturion, 2015-07-12:
I am not sure to understand. I have done this:

    cd /usr/src/linux-headers-4.1.1-cubieboard2
    wget https://github.com/igorpecovnik/lib/blob/next/patch/headers-debian-byteshift.patch
    patch -p1 headers-debian-byteshift.patch

But nothing happens...

If I try:

    patch -p1 headers-debian-byteshift.patch .

I have:

    patch: **** read error : Is a directory

berturion, 2015-07-12:
I realized that wget download the git html page of the page so I manually put the raw content of the patch in a file.
But the result of the patch command is the same, nothing happens... I don't understand.

berturion, 2015-07-12:
Ok, sorry, I just googled a bit and saw that the command to apply a path is:

    patch -p1 < headers-debian-byteshift.patch

The `<` was missing.

igorpecovnik, 2015-07-12:
You made a mistake on patch downloading ;) RAW download not HTML.

    wget https://raw.githubusercontent.com/igorpecovnik/lib/next/patching.sh
    https://raw.githubusercontent.com/igorpecovnik/lib/next/patching.sh
    make scripts

The result:

```javascript
 HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/dtc/dtc
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/modpost.o
  CC      scripts/mod/devicetable-offsets.s
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTLD  scripts/mod/modpost
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/conmakehash
  HOSTCC  scripts/sortextable
```

berturion, 2015-07-12:
No problem, thanks a lot. Compilation is ok now.

Sorry for newbie questions but, I have another problem now. I try to compile linux kernel modules for my Sony PlayTV witch uses firmware "/lib/firmware/dvb-usb-dib0700-1.20.fw".

So following instructions here http://www.linuxtv.org/wiki/index.php/How_to_Obtain,_Build_and_Install_V4L-DVB_Device_Drivers I was able to build them for my first kernel 4.0.5. But if I run again the build script, now that I have a 4.1.1 kernel, it does it as if a had the old one. So the modules are built for kernel 4.0.5.

I don't know how to specify that I am now using 4.1.1 kernel.

I installed all the .deb packages that your building script generated and ran "make scripts" from the new kernel headers sources and reboot.

Maybe I miss something but I don't know what.

igorpecovnik, 2015-07-12:
According to that script you need to specify where are your modules. 

    /lib/modules/[your kernel version]/kernel/drivers/media

they are here:

    /lib/modules/4.1.1-cubieboard2

berturion, 2015-07-12:
Ok, I had to run this before executing again there build script:

    make distclean

Then:

    ./build

Now the building process identifies my kernel version correctly :)

closed 2015-07-12 by berturion

junaidjavaid, 2017-11-04:
scripts/Makefile.host:101: recipe for target 'scripts/sign-file' failed
make[1]: *** [scripts/sign-file] Error 1
[image]
i just took start up with linux and facing this problem can you help me

igorpecovnik, 2017-11-04:
We help project related issues. This is not one of them. You can try to seek help here: https://forum.armbian.com/forum/25-peer-to-peer-technical-support/