armbian/build #77: Build with the default settings fails Cubietruck-1.4-wheezy

bofm, 2015-07-18:
It fails with the default settings. I used the compile.sh from README.md
In the menu I choose `cubietruck - kernel 3.4 - wheezy`

```bash
root@vagrant-ubuntu-trusty-64:~/ct2# ./compile.sh
./compile.sh: line 1: nd: command not found
Cloning into 'lib'...
remote: Counting objects: 1747, done.
remote: Total 1747 (delta 0), reused 0 (delta 0), pack-reused 1747
Receiving objects: 100% (1747/1747), 25.38 MiB | 3.95 MiB/s, done.
Resolving deltas: 100% (1085/1085), done.
Checking connectivity... done.
[ ok ] Building Cubietruck Debian 4.1 wheezy default
[ ok ] Syncing clock
[ ok ] Downloading u-boot
Cloning into '/home/vagrant/ct2/output/u-boot'...
remote: Counting objects: 329480, done.
remote: Compressing objects: 100% (64356/64356), done.
remote: Total 329480 (delta 263798), reused 324682 (delta 259169)
Receiving objects: 100% (329480/329480), 74.14 MiB | 7.00 MiB/s, done.
Resolving deltas: 100% (263798/263798), done.
Checking connectivity... done.
[ ok ] Downloading linux-sunxi
Cloning into '/home/vagrant/ct2/output/linux-sunxi'...
remote: Counting objects: 4116428, done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 4116428 (delta 1), reused 0 (delta 0), pack-reused 4116419
Receiving objects: 100% (4116428/4116428), 1.13 GiB | 638.00 KiB/s, done.
Resolving deltas: 100% (3427455/3427455), done.
Checking connectivity... done.
Checking out files: 100% (43235/43235), done.
[ ok ] Downloading sunxi-tools                               
Cloning into '/home/vagrant/ct2/output/sunxi-tools'...
remote: Counting objects: 1113, done.
remote: Total 1113 (delta 0), reused 0 (delta 0), pack-reused 1113
Receiving objects: 100% (1113/1113), 391.51 KiB | 526.00 KiB/s, done.
Resolving deltas: 100% (599/599), done.
Checking connectivity... done.
[ ok ] Downloading rt8192cu
Cloning into '/home/vagrant/ct2/output/rt8192cu'...
remote: Counting objects: 585, done.
remote: Total 585 (delta 0), reused 0 (delta 0), pack-reused 585
Receiving objects: 100% (585/585), 10.15 MiB | 3.45 MiB/s, done.
Resolving deltas: 100% (258/258), done.
Checking connectivity... done.
[ ok ] Downloading linux-sunxi/drivers/video/fbtft
Cloning into '/home/vagrant/ct2/output/linux-sunxi/drivers/video/fbtft'...
remote: Counting objects: 1170, done.
remote: Total 1170 (delta 0), reused 0 (delta 0), pack-reused 1170
Receiving objects: 100% (1170/1170), 526.91 KiB | 499.00 KiB/s, done.
Resolving deltas: 100% (725/725), done.
Checking connectivity... done.
[ ok ] Downloading sunxi-display-changer
Cloning into '/home/vagrant/ct2/output/sunxi-display-changer'...
remote: Counting objects: 87, done.
remote: Total 87 (delta 0), reused 0 (delta 0), pack-reused 87
Unpacking objects: 100% (87/87), done.
Checking connectivity... done.
[ ok ] Compiling sunxi tools
[ ok ] Patching kernel v4.1.1
patching file drivers/spi/Kconfig
patching file drivers/spi/Makefile
patching file drivers/spi/spi-sun7i.c
[ ok ] Compiling sunxi tools
[ ok ] Patching U-boot v2015.04
patching file configs/Lamobo_R1_defconfig
patching file configs/Awsom_defconfig
patching file include/config_distro_bootcmd.h
Hunk #1 succeeded at 173 (offset -13 lines).
Hunk #2 succeeded at 233 (offset -13 lines).
patching file include/configs/sunxi-common.h
[ ok ] Compiling universal boot loader
make: arm-linux-gnueabihf-gcc: Command not found
/bin/sh: 1: arm-linux-gnueabihf-gcc: not found
dirname: missing operand
Try 'dirname --help' for more information.
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
/home/vagrant/ct2/lib/common.sh: line 81: /home/vagrant/ct2/output/u-boot/spl/.config: No such file or directory
/home/vagrant/ct2/lib/common.sh: line 83: /home/vagrant/ct2/output/u-boot/spl/.config: No such file or directory
make: arm-linux-gnueabihf-gcc: Command not found
/bin/sh: 1: arm-linux-gnueabihf-gcc: not found
dirname: missing operand
Try 'dirname --help' for more information.
scripts/kconfig/conf --silentoldconfig Kconfig
.config:397:warning: override: reassigning to symbol ARMV7_BOOT_SEC_DEFAULT
.config:398:warning: override: reassigning to symbol OLD_SUNXI_KERNEL_COMPAT
#
# configuration written to .config
#
  CHK     include/config.h
  UPD     include/config.h
  GEN     include/autoconf.mk
/bin/sh: 1: arm-linux-gnueabihf-gcc: not found
  GEN     include/autoconf.mk.dep
/bin/sh: 1: arm-linux-gnueabihf-gcc: not found
make[1]: *** [include/autoconf.mk] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [include/autoconf.mk.dep] Error 1
  GEN     spl/include/autoconf.mk
/bin/sh: 1: arm-linux-gnueabihf-gcc: not found
make[1]: *** [spl/include/autoconf.mk] Error 1
make: *** No rule to make target `include/config/auto.conf', needed by `include/config/uboot.release'.  Stop.
cp: cannot stat ‘u-boot-sunxi-with-spl.bin’: No such file or directory
dpkg-deb: building package `linux-u-boot-3.4.108-cubietruck' in `linux-u-boot-3.4.108-cubietruck_4.1_armhf.deb'.
[ Error ] Building failed, check configuration.

```

igorpecovnik, 2015-07-18:
It looks like the packages for host are not installed by script. What if you manually install them?

https://github.com/igorpecovnik/lib#creating-compile-environment

than reboot and run the script again.

bofm, 2015-07-19:
It worked after I manually installed the packages.

Let me share my opinion.
Having a `build.sh` script instead of a snippet inside the README, as it used to be some time before, is better.

The current version of README is overloaded. It is not clear for the first look, whether the things described there are done automatically or must be done manually.

Once again, thanks a lot for the project.

igorpecovnik, 2015-07-20:
I tested on clean Ubuntu 14.04.2 LTS net install with SSH server only install. My script works out of the box. No idea what can be wrong on your system that install fails.

Why I solved this way? Enduser doesn't need to download anything and doesn't need to be faimiliar with git. Next. Call script should be user configurable while script not. Script, if left alone, is auto updating. For expert users all this is anyway irelevant.

Decision to put things this way have some sence but yes I agree it's now a little too much things in README. This part should be move to separate page / wiki / developer manual.

Thanks.

igorpecovnik, 2016-02-05:
This part has been change since much and it's performing good.

closed 2016-02-05 by igorpecovnik