armbian/build #79: [RfE] Include sunxi_tp_temp

ThomasKaiser, 2015-07-19:
Since the thermal sensor support for A10/A20 in both 4.0 and especially 4.1 is broken (values to low in 4.0 and totally weird with 4.1) it would be good to include sunxi_tp_temp in /usr/local/bin/ http://www.lemaker.org/forum.php?mod=viewthread&tid=8137&page=4#pid83281

And maybe provide a function called soctemp that calls "/usr/local/bin/sunxi_tp_temp 1447"

igorpecovnik, 2015-07-20:
I think this should be solved within kernel. I can't check right know but I think numbers were ok in last build (4.1.2) ? I can provide this alternative solution within next build.

ThomasKaiser, 2015-07-22:
I agree but do also believe kernel hackers simply don't care whether the values reported have any relationship with reality or not ;-)

I will open a new thread on the linux-sunxi list :-)

ThomasKaiser, 2016-06-07:
Fixed with https://github.com/igorpecovnik/lib/commit/5365760e0213683741486a1e4e24941fe51597cc#diff-dc1b3636afce3fde659ab59ae1a61c7c since now `/usr/local/bin/sunxi_tp_temp` is included

As a potential enhancement see [post 13](http://forum.armbian.com/index.php/topic/155-testers-wanted-sunxi-adjustments-for-rpi-monitor/?p=967) since with minor adoptions the tool could be useful with A10 and A13 also (but then SoC identification should be included into the C source and output should be in the usual range: 41.2°C --> 41200).

But since the thermal readouts from these SoCs are inaccurate anyway (see starting with post 67 in the aforementioned thread in Armbian forum) we can close the issue and leave this here as a reference.

closed 2016-06-07 by ThomasKaiser