armbian/build #84: Regarding customizing u-boot to support verified boot

ramki982, 2015-08-01:
Could you point me the shell scripts in the lib that will let me customise the u-boot output to one that supports verified boot

igorpecovnik, 2015-08-01:
Check script configuration.sh ... there are definitions for U-boot sources. For mainstream u-boot source, you can change tag in call script. Currently it's set for 2015.04, latest-latest stable. Currently it's 2015.07

I only download source, patch it for minor changes (patching.sh) and compile (common.sh).

ramki982, 2015-08-02:
Thanks a bunch. I have to add some CONFIGs to lamboro def config before u boot is compiled. Let me check if these can b appended to that file at runtime thro simple echo before u boot starts to compile

ramki982, 2015-08-04:
I managed to customise the boards.sh to create a FIT IMAGE format file. I try to use the commands in u-boot prompt to boot it

setenv bootargs console=ttyS0,115200 sunxi_g2d_mem_reserve=0 sunxi_ve_mem_reserve=0 root=/dev/mmcblk0p1 rootfstype=ext4 elevator=deadline rootwait
ext4load mmc 0 0x43000000 /boot/script.bin
ext4load mmc 0 0x48000000 /boot/fitImage
bootm 0x48000000

Kernel seems to start booting but after some time - I get a kernel panic 
VFS: Cannot open root device "mmc0p1" or unknown-block(0,0): error -6
Please append a correct "root=" boot option; here are the available partitions:
<0>Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[<c0015018>] (unwind_backtrace+0x0/0x134) from [<c0704ad4>] (panic+0xa0/0x1d4)
[<c0704ad4>] (panic+0xa0/0x1d4) from [<c09d2dc0>] (mount_block_root+0x194/0x224)
[<c09d2dc0>] (mount_block_root+0x194/0x224) from [<c09d3044>] (mount_root+0x108/0x110)
(mount_root+0x108/0x110)
[<c09d3044>] (mount_root+0x108/0x110) from [<c09d31dc>] (prepare_namespace+0x190/0x1d8)
(prepare_namespace+0x190/0x1d8)
[<c09d31dc>] (prepare_namespace+0x190/0x1d8) from [<c09d2a94>] (kernel_init+0x240/0x248)
(kernel_init+0x240/0x248)
[<c09d2a94>] (kernel_init+0x240/0x248) from [<c000f478>] (kernel_thread_exit+0x0/0x8)

Do you see any clues why this would happen.

thanks
Ram

igorpecovnik, 2015-08-04:
Hmm, not exactly. Maybe the environment declaration is somehow invalid / not accepted? Try with minimum: setenv bootargs console=ttyS0,115200 root=/dev/mmcblk0p1

ramki982, 2015-08-04:
Even though i pass the root as /dev/mmcblk0p1 using u-boot bootargs, when the kernel boots, it seems to get it as mmc0p1. I think i'm hitting http://forums.xilinx.com/t5/Embedded-Linux/U-Boot-not-passing-bootargs-to-Linux/td-p/343883

As a trial - i've set the CONFIG_CMDLINE in linux-sunxi.config as below
CONFIG_CMDLINE="console=ttyS0,115200 sunxi_g2d_mem_reserve=0 sunxi_ve_mem_reserve=0 root=/dev/mmcblk0p1 rootfstype=ext4 elevator=deadline rootwait"

As this is the value that I expected u-boot to send to the kernel.

I'd check if this solves the issue.

thanks

ramki982, 2015-08-07:
I did solve this issue for now by updating the linux-sunxi.config in lib/config 

CONFIG_CMDLINE="console=ttyS0,115200 sunxi_g2d_mem_reserve=0 sunxi_ve_mem_reserve=0 root=/dev/mmcblk0p1 rootfstype=ext4 elevator=deadline rootwait"

With this - I'm able to boot using FIT IMAGE & use Verified UBOOT

closed 2016-07-25 by zador-blood-stained