armbian/build #85: Add utopic and vivid for Ubuntu options

blindpet, 2015-08-02:
Is this planned or is there a stability/compatibility issue?

igorpecovnik, 2015-08-02:
Adding Vivid is low priority and doesn't cover time and troubles. I also don't see much advantages since Debian Jessie is on similar technological level and in much better shape. Working with very changing basis is usually a nightmare :)

blindpet, 2015-08-02:
I hear you, I figured it could be there for testing maybe in a separate branch so you don't get issue tickets relating to vivid?

I don't use it myself but I know many people do or will want to and your images are the best :)

igorpecovnik, 2015-08-02:
I remember how much problems I had working with, in that time very changing, Jessie ... I actually spent a day trying to bring Vivid into script but quickly realized that this is far from easy to meet the quality level of Wheezy / Jessie / Trusty. And this is going away from real problems ...

blindpet, 2015-08-02:
I hear you, working on updating my version of the installer script to support systemd and it is pissing me off a lot. Users think latest and greatest is best but EOL isn't for 3 years or so for jessie and 14.04, utopic seems reasonable to include though doesn't it?

igorpecovnik, 2015-08-03:
It make sense but as I said - not much real advantages. Despite that I might add it someday when I feel my self very boring ;) ... or if someone else does the job.

closed 2015-09-11 by igorpecovnik