armbian/build #86: does current scripts work for 4.1.2?

tsgan, 2015-08-06:
Checking connectivity... done.
[ ok ] Compiling sunxi tools
[ ok ] Patching kernel v4.1.2
/home/tsgan/armbian/lib/patching.sh: line 20: cd: /home/tsgan/armbian/output/linux-mainline: No such file or directory
error: pathspec 'v4.1.2' did not match any file(s) known to git.
cat: drivers/net/wireless/brcm80211/brcmfmac/feature.c: No such file or directory
sed: can't read drivers/net/wireless/brcm80211/brcmfmac/feature.c: No such file or directory
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/arch/arm/boot/dts/sun7i-a20-cubietruck.dts
|+++ b/arch/arm/boot/dts/sun7i-a20-cubietruck.dts
--------------------------
File to patch:

igorpecovnik, 2015-08-06:
Huh. It should ... this patching part is critical in any case since patches can't patch all sources with one patch. I am about to update the script in few days.

tsgan, 2015-08-06:
More complete log:

```
/home/tsgan/armbian/lib/common.sh: line 255: cd: /home/tsgan/armbian/output/output/kernel/: No such file or directory
ls: cannot access *next*.tar: No such file or directory
[ ok ] Compiling kernel
[ ok ] Syncing clock
[ ok ] Downloading u-boot
Already up-to-date.
[ ok ] Downloading linux-mainline
Cloning into '/home/tsgan/armbian/output/linux-mainline'...
remote: Counting objects: 4624114, done.
remote: Compressing objects: 100% (358009/358009), done.
error: index-pack died of signal 9624114), 723.67 MiB | 326.00 KiB/s
fatal: index-pack failed
[ ok ] Downloading sunxi-tools
Already up-to-date.
[ ok ] Downloading rt8192cu
Already up-to-date.
[ ok ] Downloading sunxi-display-changer
Already up-to-date.
[ ok ] Compiling sunxi tools
[ ok ] Patching kernel v4.1.2
/home/tsgan/armbian/lib/patching.sh: line 20: cd: /home/tsgan/armbian/output/linux-mainline: No such file or directory
error: pathspec 'v4.1.2' did not match any file(s) known to git.
cat: drivers/net/wireless/brcm80211/brcmfmac/feature.c: No such file or directory
sed: can't read drivers/net/wireless/brcm80211/brcmfmac/feature.c: No such file or directory
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/arch/arm/boot/dts/sun7i-a20-cubietruck.dts
|+++ b/arch/arm/boot/dts/sun7i-a20-cubietruck.dts
--------------------------
File to patch:
```

igorpecovnik, 2015-08-06:
Cloning into '/home/tsgan/armbian/output/linux-mainline'...
    remote: Counting objects: 4624114, done.
    remote: Compressing objects: 100% (358009/358009), done.
    error: index-pack died of signal 9624114), 723.67 MiB | 326.00 KiB/s
    fatal: index-pack failed

This is a network error. Sources were not downloaded successfully.

tsgan, 2015-08-06:
oh, right. I will try.

tsgan, 2015-08-06:
Was able to get linux.
Some error related to patching:
```
[ ok ] Patching kernel v4.1.2
patching file arch/arm/boot/dts/sun7i-a20-cubietruck.dts
Hunk #1 FAILED at 71.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm/boot/dts/sun7i-a20-cubietruck.dts.rej
patching file arch/arm/boot/dts/sun7i-a20-bananapro.dts
patch unexpectedly ends in middle of line
Hunk #1 FAILED at 159.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm/boot/dts/sun7i-a20-bananapro.dts.rej
patching file scripts/Makefile.dtbinst
patching file scripts/package/builddeb
Hunk #6 succeeded at 363 (offset 11 lines).
patching file arch/arm/kernel/asm-offsets.c
Reversed (or previously applied) patch detected!  Assuming -R.
Hunk #2 succeeded at 40 (offset 1 line).
[ ok ] Patching U-boot v2015.04
patching file configs/Lamobo_R1_defconfig
patching file configs/Awsom_defconfig
patching file include/config_distro_bootcmd.h
Hunk #1 succeeded at 173 (offset -13 lines).
Hunk #2 succeeded at 233 (offset -13 lines).
patching file include/configs/sunxi-common.h
```

tsgan, 2015-08-06:
Some more messages:
```
Processing triggers for systemd (215-17+deb8u1) ...
[ ok ] Closing and preparing cache
[ ok ] Install kernel
tar: unrecognized option '--msgbox'
Try 'tar --help' or 'tar --usage' for more information.
/home/tsgan/armbian/lib/boards.sh: line 188: cd: /home/tsgan/armbian/output/output/sdcard/usr/src/linux-headers-Box: No such file or directory
/home/tsgan/armbian/lib/boards.sh: line 190: [: too many arguments
tar: unrecognized option '--msgbox'
Try 'tar --help' or 'tar --usage' for more information.
[ ok ] Fixing release custom applications
```

tsgan, 2015-08-06:
More messages:
```
make[1]: Entering directory `/home/tsgan/armbian/output/linux-mainline'
make[1]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/tsgan/armbian/output/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/driver.o
In file included from <command-line>:0:0:
././include/linux/kconfig.h:4:32: fatal error: generated/autoconf.h: No such file or directory
 #include <generated/autoconf.h>
                                ^
compilation terminated.
make[2]: *** [/home/tsgan/armbian/output/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/driver.o] Error 1
make[1]: *** [_module_/home/tsgan/armbian/output/usb-redirector-linux-arm-eabi/files/modules/src/tusbd] Error 2
make[1]: Leaving directory `/home/tsgan/armbian/output/linux-mainline'
make: *** [default] Error 2
cp: cannot stat ‘/home/tsgan/armbian/output/usb-redirector-linux-arm-eabi/files/modules/src/tusbd/tusbd.ko’: No such file or directory
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /home/tsgan/armbian/output/linux-mainline/ M=/home/tsgan/armbian/output/rt8192cu  modules
make[1]: Entering directory `/home/tsgan/armbian/output/linux-mainline'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  Building modules, stage 2.
./scripts/Makefile.modpost:42: include/config/auto.conf: No such file or directory
make[2]: *** No rule to make target `include/config/auto.conf'.  Stop.
make[1]: *** [modules] Error 2
make[1]: Leaving directory `/home/tsgan/armbian/output/linux-mainline'
make: *** [modules] Error 2
cp: cannot stat ‘*.ko’: No such file or directory
[ ok ] Fingerprinting
[ ok ] Possible after install
[ ok ] Writing boot loader
[ ok ] Shrink partition and image to real size with 10% free space
resize2fs 1.42.9 (4-Feb-2014)
[ ok ] Create and sign download ready ZIP archive
  adding: Lamobo-r1_Debian_3.2_jessie_4.1.2.raw (deflated 61%)
  adding: readme.txt (deflated 62%)
  adding: imagewriter.exe (deflated 36%)
[ ok ] Runtime 33 min
tsgan@tsgan-virtual-machine:~/armbian$
```

igorpecovnik, 2015-08-06:
The patching part is not critical. You should be able to build even if patching is not succeeded. 
BTW. What is your host system? Ubuntu 14 LTS is recommended since others might bring more troubles.

tsgan, 2015-08-06:
Figured it out.

closed 2015-08-06 by tsgan