armbian/build #87: U-boot compilation errors for Lamobo R1

ramki982, 2015-08-08:
Hi Igor,

Sincere thanks for all your incredible work. One observation I have had over last week while using the ARMBIAN scripts with Lamobo R1 board.

Unless I add the following - I get a bunch of compilation errors - Could you help add this to the PATCH
# CONFIG_CMD_IMLS is not set

-------errors--------------
In file included from common/cmd_bootm.c:35:0:
include/mtd/cfi_flash.h:168:22: error: unknown type name ‘flash_info_t’
 void flash_write_cmd(flash_info_t * info, flash_sect_t sect,
                      ^
include/mtd/cfi_flash.h:168:43: error: unknown type name ‘flash_sect_t’
 void flash_write_cmd(flash_info_t * info, flash_sect_t sect,
                                           ^
include/mtd/cfi_flash.h:172:22: error: unknown type name ‘flash_info_t’
 void flash_cmd_reset(flash_info_t *info);
                      ^
common/cmd_bootm.c:36:1: error: unknown type name ‘flash_info_t’
 extern flash_info_t flash_info[]; /* info for FLASH chips */
 ^
common/cmd_bootm.c: In function ‘do_imls_nor’:
common/cmd_bootm.c:321:2: error: unknown type name ‘flash_info_t’
  flash_info_t *info;
  ^
common/cmd_bootm.c:326:7: error: ‘CONFIG_SYS_MAX_FLASH_BANKS’ undeclared (first use in this function)
   i < CONFIG_SYS_MAX_FLASH_BANKS; ++i, ++info) {
       ^
common/cmd_bootm.c:326:7: note: each undeclared identifier is reported only once for each function it appears in
common/cmd_bootm.c:328:11: error: request for member ‘flash_id’ in something not a structure or union
   if (info->flash_id == FLASH_UNKNOWN)
           ^
common/cmd_bootm.c:328:25: error: ‘FLASH_UNKNOWN’ undeclared (first use in this function)
   if (info->flash_id == FLASH_UNKNOWN)
                         ^
common/cmd_bootm.c:330:23: error: request for member ‘sector_count’ in something not a structure or union
   for (j = 0; j < info->sector_count; ++j) {
                       ^
common/cmd_bootm.c:332:22: error: request for member ‘start’ in something not a structure or union
    hdr = (void *)info->start[j];
                      ^
make[1]: *** [common/cmd_bootm.o] Error 1
make: *** [common] Error 2
----------------------------

After adding that line, when I make - I get the following errors
------------errors-----------------
Device Tree Source is not correctly specified.
Please define 'CONFIG_DEFAULT_DEVICE_TREE'
or build with 'DEVICE_TREE=<device_tree>' argument
------------------------------------
To fix that I also need to add
CONFIG_DEFAULT_DEVICE_TREE="sun7i-a20-bananapi"

So in essence to get the compilation of u-boot for Lamobo r1 - I'm having to add 2 lines 
# CONFIG_CMD_IMLS is not set
CONFIG_DEFAULT_DEVICE_TREE="sun7i-a20-bananapi"

Wanted to share these observations with you so that they can be made part of the patch

igorpecovnik, 2015-08-08:
Which U-boot are you using? It's not working with recent one - 2015.07 ... I already try to fix but it looks like it's not trivial.

ramki982, 2015-08-08:
I used 
```
git clone git://git.denx.de/u-boot.git 
```
so it should be u-boot 2015.07 - 
I'm able to successfully build it for Lamobo R1 with these 2 lines

ramki982, 2015-08-11:
Some more updates, 
Based on suggestion from a teammate - i copied the sun7i-a20-lamobo-r1.dts to u-boot/arch/arm/dts folder. Made a minor fix to that file (replace ir0_pins_a   with ir0_rx_pins_a)

And included that sun7i-a20-lamobo-r1.dtb in makefile in the dts folder.

With the above changes in place - I was also able to compile u-boot by using the following default tree
CONFIG_DEFAULT_DEVICE_TREE="sun7i-a20-lamobo-r1"

After this when I checked the version of mkimage - I got the following
mkimage version 2014.10+dfsg1-3

rpodgorny, 2015-10-30:
@ramki982 where did you get sun7i-a20-lamobo-r1.dts from?

ramki982, 2015-10-30:
https://github.com/igorpecovnik/lib/blob/next/patch/sun7i-a20-lamobo-r1.dts

It is in the ARMBIAN git

Thx
Ram

igorpecovnik, 2015-11-12:
It's compiling fine now with 2015.10

closed 2015-11-12 by igorpecovnik