armbian/build #88: issue compiling next-MICRO

Sunderfield, 2015-08-12:
have tried to recompile with new pulls but getting errors installing kernel
```
  INSTALL debian/headertmp/usr/include/linux/spi/ (1 file)
  INSTALL debian/headertmp/usr/include/linux/sunrpc/ (1 file)
  INSTALL debian/headertmp/usr/include/linux/tc_act/ (11 files)
  INSTALL debian/headertmp/usr/include/linux/tc_ematch/ (4 files)
  INSTALL debian/headertmp/usr/include/linux/usb/ (11 files)
  INSTALL debian/headertmp/usr/include/linux/wimax/ (1 file)
  INSTALL debian/headertmp/usr/include/linux/ (413 files)
  INSTALL debian/headertmp/usr/include/misc/ (1 file)
  INSTALL debian/headertmp/usr/include/mtd/ (5 files)
  INSTALL debian/headertmp/usr/include/rdma/ (6 files)
  INSTALL debian/headertmp/usr/include/scsi/fc/ (4 files)
  INSTALL debian/headertmp/usr/include/scsi/ (3 files)
  INSTALL debian/headertmp/usr/include/sound/ (11 files)
  INSTALL debian/headertmp/usr/include/video/ (3 files)
  INSTALL debian/headertmp/usr/include/xen/ (4 files)
  INSTALL debian/headertmp/usr/include/uapi/ (0 file)
  INSTALL debian/headertmp/usr/include/asm/ (34 files)
dpkg-deb: building package `linux-firmware-image-4.1.4-rt1-micro' in `../linux-f
irmware-image-4.1.4-rt1-micro_1.9_armhf.deb'.
dpkg-deb: building package `linux-dtb-4.1.4-rt1-micro' in `../linux-dtb-4.1.4-rt
1-micro_1.9_armhf.deb'.
dpkg-deb: building package `linux-headers-4.1.4-rt1-micro' in `../linux-headers-
4.1.4-rt1-micro_1.9_armhf.deb'.
dpkg-deb: building package `linux-libc-dev' in `../linux-libc-dev_1.9_armhf.deb'
.
dpkg-deb: building package `linux-image-4.1.4-rt1-micro' in `../linux-image-4.1.
4-rt1-micro_1.9_armhf.deb'.
[ ok ] Creating 1500 Mb SD card image with 1 Mb reservation for u-boot
[ ok ] Extracting jessie from cache. Your cache is 1 days old.
[ ok ] Install kernel
sed: can't read /home/jzonneveld/kernel3/output/output/sdcard/etc/init.d/cpufreq
utils: No such file or directory
sed: can't read /home/jzonneveld/kernel3/output/output/sdcard/etc/init.d/cpufreq
utils: No such file or directory
sed: can't read /home/jzonneveld/kernel3/output/output/sdcard/etc/init.d/cpufreq
utils: No such file or directory
/home/jzonneveld/kernel3/lib/boards.sh: line 151: /home/jzonneveld/kernel3/outpu
t/output/sdcard/etc/hostname: No such file or directory
/home/jzonneveld/kernel3/lib/boards.sh: line 154: /home/jzonneveld/kernel3/outpu
t/output/sdcard/etc/hosts: No such file or directory
cp: target /home/jzonneveld/kernel3/output/output/sdcard/etc/network/ is not a
 directory
ln: failed to create symbolic link /home/jzonneveld/kernel3/output/output/sdcar
d/etc/network/interfaces: No such file or directory
mount: mount point /home/jzonneveld/kernel3/output/output/sdcard/tmp does not ex
ist
chroot: failed to run command /bin/bash: No such file or directory
chroot: failed to run command /bin/bash: No such file or directory
chroot: failed to run command /bin/bash: No such file or directory
/home/jzonneveld/kernel3/lib/boards.sh: line 188: cd: /home/jzonneveld/kernel3/o
utput/output/sdcard/usr/src/linux-headers-4.1.4-micro: No such file or directory
cp: cannot create regular file /home/jzonneveld/kernel3/output/output/sdcard/bo
ot/boot.cmd: No such file or directory
chroot: failed to run command /bin/bash: No such file or directory
mkimage: Can't open /home/jzonneveld/kernel3/output/output/sdcard/boot/boot.scr:
 No such file or directory
cp: cannot create regular file /home/jzonneveld/kernel3/output/output/sdcard/bo
ot/boot.bmp: No such file or directory
checkdir:  cannot create extraction directory: /home/jzonneveld/kernel3/output/o
utput/sdcard/lib/firmware
           No such file or directory
[ ok ] Fixing release custom applications
cp: cannot create regular file /home/jzonneveld/kernel3/output/output/sdcard/et
c/init/ttyS0.conf: No such file or directory
cp: cannot stat /home/jzonneveld/kernel3/output/output/sdcard/lib/systemd/syste
m/serial-getty@.service: No such file or directory
sed: can't read /home/jzonneveld/kernel3/output/output/sdcard/etc/systemd/system
/getty.target.wants/serial-getty@ttyS0.service: No such file or directory
[... 61 lines cut ...]
t/output/sdcard/root/readme.txt: No such file or directory
/home/jzonneveld/kernel3/lib/common.sh: line 344: /home/jzonneveld/kernel3/outpu
t/output/sdcard/root/readme.txt: No such file or directory
/home/jzonneveld/kernel3/lib/common.sh: line 345: /home/jzonneveld/kernel3/outpu
t/output/sdcard/root/readme.txt: No such file or directory
/home/jzonneveld/kernel3/lib/common.sh: line 346: /home/jzonneveld/kernel3/outpu
t/output/sdcard/root/readme.txt: No such file or directory
/home/jzonneveld/kernel3/lib/common.sh: line 347: /home/jzonneveld/kernel3/outpu
t/output/sdcard/root/readme.txt: No such file or directory
/home/jzonneveld/kernel3/lib/common.sh: line 348: /home/jzonneveld/kernel3/outpu
t/output/sdcard/root/readme.txt: No such file or directory
[ ok ] Possible after install
chroot: failed to run command /bin/bash: No such file or directory
chroot: failed to run command /bin/bash: No such file or directory
umount: /home/jzonneveld/kernel3/output/output/sdcard/tmp: mountpoint not found
cp: cannot stat /home/jzonneveld/kernel3/output/output/sdcard/root/readme.txt:
 No such file or directory
rm: cannot remove /home/jzonneveld/kernel3/output/output/sdcard/usr/bin/qemu-ar
m-static: No such file or directory
[ ok ] Writing boot loader
[ ok ] Shrink partition and image to real size with 10% free space
/home/jzonneveld/kernel3/lib/common.sh: line 363: *512: syntax error: operand ex
pected (error token is "*512")
[ ok ] Runtime 31 min
```

igorpecovnik, 2015-08-12:
Your root filesystem looks corrupted.

reboot & try again

If failed, remove /output/output/rootfs/jessie.tgz
reboot
& try again.

davidahoward, 2015-08-12:
I get this happening with 'next' either debian jessie or ubuntu trusty

[ ok ] Writing boot loader
[ ok ] Shrink partition and image to real size with 10% free space
/home/dhoward/Desktop/Lime2/lib/common.sh: line 363: *512: syntax error: operand expected (error token is "*512")
[ ok ] Runtime 78 min

davidahoward, 2015-08-13:
I found that there was a problem in common.sh, 

# partition prepare

LOOP=$(losetup -f)
losetup $LOOP $RAWIMAGE
PARTSTART=$(fdisk -l $LOOP | tail -1 | awk '{ print $2}')
PARTSTART=$(($PARTSTART*512))

I changed tail -1 to tail -2 and it correctly got the partition start...  
tail -1 gets a blank line then errors out with

[ ok ] Shrink partition and image to real size with 10% free space
/home/dhoward/Desktop/Lime2/lib/common.sh: line 363: *512: syntax error: operand expected (error token is "*512")

davidahoward, 2015-08-13:
also needs to be fixed in second part

# truncate the image
TRUNCATE=$(fdisk -l $LOOP | tail -2 | awk '{ print $3}')
TRUNCATE=$((($TRUNCATE+1)*512))

Sunderfield, 2015-08-13:
Testing it now.

Sunderfield, 2015-08-13:
next error:

```
[ ok ] Install kernel
/home/jzonneveld/kernel3/lib/boards.sh: line 188: cd: /home/jzonneveld/kernel3/output/output/sdcard/usr/src/linux-headers-4.1.1-micro: No such file or directory
[ ok ] Fixing release custom applications
```

it did however build them: 
```
dpkg-deb: building package `linux-headers-4.1.1-rt1-micro' in `../linux-headers-4.1.1-rt1-micro_1.9_armhf.deb'.
```

igorpecovnik, 2015-08-13:
OK, I am going to fix / hardening this part. 10x

Sunderfield, 2015-08-13:
I just have a bit too little confidence in my skills to fix this

igorpecovnik, 2015-08-13:
@davidahoward 

This code is o.k. but it fails if you run out of space or partition creation is not successful.

Are you using Ubuntu Trusty?

igorpecovnik, 2015-08-13:
@Sunderfield 

I think it's related to the kernel naming. Not sure at the moment but those errors should be out of next major update.

davidahoward, 2015-08-18:
my build environment is a VM under OSX..

lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.10
Release:	14.10
Codename:	utopic

igorpecovnik, 2015-09-11:
On host system different than Ubuntu 14.04 LTS troubles are expected.

closed 2015-09-11 by igorpecovnik