armbian/build #89: Shrink Image error

blindpet, 2015-08-14:
Just did a fresh image build today and got this error:

    [ ok ] Shrink partition and image to real size with 10% free space
/home/htpcguides/lib/common.sh: line 363: *512: syntax error: operand expected (error token is "*512")

igorpecovnik, 2015-08-15:
This error usually happens if you set too small initial image for the stuff you install. I am closing on an large update and this should be fixed. Because of automatic shrinking it's pointless to have an option to set initial image in basic config. If I set an initial image to 4GB it's enough for 99% of scenarios. For the rest 1% there will be an error "please increase your initial SD size ... " :)

davidahoward, 2015-08-19:
Ive seen this problem due to the fdisk command returning a blank line and script assuming that the last line would have the size information.

igorpecovnik, 2015-08-20:
Huh. It looks like fdisk output is different on some systems. This should be more resilient:

    PARTSTART=$(fdisk -l $LOOP | grep $LOOP | grep Linux | awk '{ print $2}')

closed 2015-09-01 by igorpecovnik

blindpet, 2015-09-02:
Still getting this, did a fresh install

[ ok ] Shrink partition and image to real size with 10% free space
resize2fs 1.42.12 (29-Aug-2014)
/home/htpcguides/lib/common.sh: line 385: (4k: value too great for base (error token is "4k")
[ ok ] Runtime 30 min

igorpecovnik, 2015-09-02:
Ok, but you are not using Ubuntu 14.04 ?

blindpet, 2015-09-02:
Ah yes, using 15.04 64 bit, what's the compatibility issue?

igorpecovnik, 2015-09-02:
I need to convert kbyte to byte. Can done ... I just hate such fixes ;)

igorpecovnik, 2015-09-02:
Fixed in dev version.

https://github.com/igorpecovnik/lib/commit/4fb18ff59e3176b04c1c6b7aec54b7562176c98c