armbian/build #91: Kernel 4.2 mainline: error while compiling for bananapi

skimpax, 2015-09-01:
Compiling mainline 4.2 kernel failed due to a tdb issue (my target is a bananapi, Debian Jessie on top of it).
I just had to remove the sun7i-a20-lamobo-r1.dtb inclusion in patching.sh to make compilation succeed. Kernel 4.2 runs perfectly on my bananapi (since few hours however).

I suppose this inclusion were made for some purpose: removing it works for bananapi but wouldn't is cause side effect for some other boards? I am not able to make decision about it.

FYI, here is my fix:
```bash
--- patching.sh.orig	2015-09-01 14:46:06.047576975 +0200
+++ patching.sh	2015-09-01 21:24:11.355086469 +0200
@@ -44,12 +44,6 @@
 		patch -p1 < $SRC/lib/patch/packaging-next.patch
 	fi
 
-	# copy bananar1 DTS
-	if [ "$(cat arch/arm/boot/dts/Makefile | grep sun7i-a20-lamobo-r1)" == "" ]; then
-		sed -i 's/sun7i-a20-bananapi.dtb \\/sun7i-a20-bananapi.dtb \\\n    sun7i-a20-lamobo-r1.dtb \\/g' arch/arm/boot/dts/Makefile
-		cp $SRC/lib/patch/sun7i-a20-lamobo-r1.dts arch/arm/boot/dts/
-	fi
-	
 	# copy orange pi DTS
 	if [ "$(cat arch/arm/boot/dts/Makefile | grep sun7i-a20-orangepi)" == "" ]; then
 		sed -i 's/sun7i-a20-bananapi.dtb \\/sun7i-a20-bananapi.dtb \\\n    sun7i-a20-orangepi.dtb \\/g' arch/arm/boot/dts/Makefile
@@ -157,4 +151,4 @@
 		patch --batch -N -p1 < $SRC/lib/patch/sunxi-boot-splash.patch
 	fi
 fi
-}
\ Pas de fin de ligne à la fin du fichier
+}
```

My kernel 4.2 compilation script:

```bash
#!/bin/bash 
# 
#   Edit and execute this script - Ubuntu 14.04 x86/64 recommended
#
#   Check https://github.com/igorpecovnik/lib for possible updates
#

# method
KERNEL_ONLY="yes"                            # build kernel only
SOURCE_COMPILE="yes"                        # force source compilation
KERNEL_CONFIGURE="no"                       # change default configuration
KERNEL_CLEAN="yes"                          # MAKE clean before compilation
USEALLCORES="yes"                           # Use all CPU cores
BUILD_DESKTOP="no"                          # desktop with hw acceleration for some boards 

# user 
DEST_LANG="en_US.UTF-8"                     # sl_SI.UTF-8, en_US.UTF-8
CONSOLE_CHAR="UTF-8"                        # console charset
TZDATA="Europe/Paris"                   # time zone
ROOTPWD="1234"                              # forced to change @first login
SDSIZE="1500"                               # SD image size in MB
AFTERINSTALL=""                             # command before closing image 
MAINTAINER="Igor Pecovnik"                  # deb signature
MAINTAINERMAIL="igor.pecovnik@****l.com"    # deb signature
GPG_PASS=""                                 # set GPG password for non-interactive packing

# advanced
KERNELTAG="v4.2"                          # kernel TAG - valid only for mainline
UBOOTTAG="v2015.07"                         # kernel TAG - valid for all sunxi
FBTFT="yes"                                 # https://github.com/notro/fbtft 
EXTERNAL="yes"                              # compile extra drivers
FORCE="yes"                                 # ignore manual changes to source

# source is where we start the script
SRC=$(pwd)
# destination
DEST=$(pwd)/output                                      
# get updates of the main build libraries
if [ -d "$SRC/lib" ]; then
    cd $SRC/lib
#    git pull 
else
    # download SDK
    apt-get -y -qq install git
    git clone https://github.com/igorpecovnik/lib
fi
source $SRC/lib/main.sh
```

igorpecovnik, 2015-09-01:
Current Lamobo R1 patch breaks U-boot compilation 2015.07 and current dev. It works v2015.04 ... I need to dig in into u-boot and prepare another patch.

igorpecovnik, 2015-09-01:
Setting UBOOTTAG="v2015.07"  to v2015.04 is also one solution. I am currently fixing new branch where I changed this part completely.

igorpecovnik, 2015-10-24:
banana compilation works.

closed 2015-10-24 by igorpecovnik