armbian/build #97: sunxi-tools cannot be copied (seem also not built)

rogierhofboer, 2015-09-09:
cp: cannot stat ‘/home/rogier/armbian/sources/sunxi-tools/fex2bin’: No such file or directory
cp: cannot stat ‘/home/rogier/armbian/sources/sunxi-tools/bin2fex’: No such file or directory
cp: cannot stat ‘/home/rogier/armbian/sources/sunxi-tools/nand-part’: No such file or directory

igorpecovnik, 2015-09-09:
They are probably not build. It looks like host is not fully compatible.

rogierhofboer, 2015-09-09:
Do you use a 64-bit or 32-bit Ubuntu?

igorpecovnik, 2015-09-09:
64bit

rogierhofboer, 2015-09-09:
It does work with Ubuntu Trusty 14.04.3 LTS 64-bit :)
(failed on Mint 17.2)

closed 2015-09-09 by rogierhofboer

deskwizard, 2015-11-21:
Issue still stands on Ubuntu Mate 14.04.2
any idea how to check what I am missing ?

deskwizard, 2015-11-21:
Cause:
libusb.h missing

Solution:
sudo apt-get install libusb-1.0-0-dev