armbian/build #103: make CONFIG_AHCI_SUNXI=m default on mainline

blindpet, 2015-09-17:
Testing port multipliers with the 4.x jessie images and CONFIG_AHCI_SUNXI is set to Y but the port multiplier isn't working. Either way it should be modular so that you can switch between port multiplier and no port multiplier without recompiling kernel.

igorpecovnik, 2015-09-17:
Than we won't be able to boot from SATA. In this case I need to add initrd which is a little more then just a config set ;)

zador-blood-stained, 2015-10-05:
I checked the source for ahci_sunxi, and if you are talking about setting enable_pmp parameter, did you try adding `ahci-sunxi.enable_pmp=1` or `ahci_sunxi.enable_pmp=1` to bootargs in boot script? This should pass parameter to builtin module, I'm only not sure about dash or underscore in module name.

igorpecovnik, 2016-05-17:
provided kernel parameter should work

closed 2016-05-17 by igorpecovnik