armbian/build #115: Onboard LED as status indicator (sdcard r/w)

dllud, 2015-10-08:
Most (all?) boards supported by Armbian have an onboard LED. It would be much handy if it could be configured to act as a system status indicator, for instance blinking on sdcard r/w.

Such visual feedback is a must when running headless (no monitor connected, expecting to establish an SSH connection).

zador-blood-stained, 2015-10-08:
It can be configured, it's just not enabled by default (mostly).
Try entering `ls /sys/class/leds/` to check what LEDs are available, 
`cat /sys/class/leds/<any led>/trigger` to see available triggers, for example
```
# cat /sys/class/leds/white:ph11:led3/trigger
[none] battery-charging-or-full battery-charging battery-full battery-charging-blink-full-solid ac-online usb-online mmc0 mmc1 timer disk-activity heartbeat backlight cpu0 cpu1 default-on rfkill0 rfkill1
```
and setting them manually, example:
```
echo "mmc0" > /sys/class/leds/blue:ph21:led1/trigger
echo "heartbeat" > /sys/class/leds/green:ph07:led4/trigger
echo "cpu0" > /sys/class/leds/orange:ph20:led2/trigger
echo "cpu1" > /sys/class/leds/white:ph11:led3/trigger
```
You need to use root shell to set triggers, using `sudo` from non-privileged user won't work.

dllud, 2015-10-08:
@zador-blood-stained Thanks for the thorough explanation. I added the echo commands to `/etc/rc.local`and it works good on Cubieboard2-2Cards. The only issue is that it only kicks in at the end of the boot process, meanwhile I get no feedback for sdcard r/w.

I feel this should be added to .fex and .dts files on Armbian's default config. The First Boot is the moment when this feedback is more useful.

zador-blood-stained, 2015-10-08:
I agree about first boot and early init, you should ask @igorpecovnik about enabling LEDs by default. Meanwhile try to set triggers in fex/bin (it's more or less self-explanatory) if you are on old kernel or install device-tree-compiler and check `man dtc` how to convert dtb to dts and vice versa, if you are using mainline ([DT Leds documentiaion](https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/leds/common.txt) )

igorpecovnik, 2015-10-08:
They are enabled by default. I disable them - change functionality. I usually have a desk full of boards and if all of them were blinking, I would go crazy. Next. I don't see much point in this except maybe at first boot ... but I did (maybe not for all boards yet) enable SD card activity LED. If it's blinking or it's on, it's most likely booting.

zador-blood-stained, 2015-10-08:
I don't think they are enabled in mainline - if they were, I would have noticed 100% CPU usage by logging processes just by looking at my cubie while testing broken Jessie image), 
I'm using cron to turn LEDs off during night because "heartbeat" trigger in dark room is annoying as hell with any LED color)

dllud, 2015-10-08:
@igorpecovnik It is disabled with mainline kernel on Cubieboard2. I think it's worth enabling one LED for SD card activity on all boards and all kernels. I really missed something like that when I had to deal with the broken jessie image.

igorpecovnik, 2015-10-09:
Aha, mainline. I agree it's cool to have at least SD activity. But It's hard to maintain those patches, even for this small change. The best would be that this is done systematically for all major boards and up-streamed. Such section need to be added (and perhaps test too) to each board:
https://github.com/igorpecovnik/lib/blob/second/patch/devices/sun7i-a20-bananapi-r1.dts#L66

It's low priority / not urgent so I'll be slow in changing this.

igorpecovnik, 2016-05-17:
disk activity LED was added to some boards

closed 2016-05-17 by igorpecovnik