armbian/build #117: Boot from any mmc on boards with multiple mmcs (mainline kernel)

dllud, 2015-10-12:
As found on #114, with mainline kernel, we get a kernel panic when trying to boot from the second card on a board with multiple active cards. This may happen on boards with multiple sdcard slots or with a combination of sdcards and emmc devices.

The origin of the kernel panic is rootfs which is set to mmcblk0 on /boot/boot.cmd

There are at least two ways of solving this:
a) Wait for sunxi mmc driver to port a feature present on the old kernel that made sure boot device always gets mmcblk0: linux-sunxi/linux-sunxi#242 (preferable?).
b) Set rootfs by uuid.

dllud, 2015-10-12:
Ok, we've got an answer. Mainline kernel sunxi mmc driver won't get any boot dev swapping. They recommend usage of rootfs by uuid.

I have just tested it on my CB2-2Card. I got the boot partition PARTUUID from `blkid`. On `boot.cmd` I set `root=PARTUUID=XXXXXXXX-XX` while *capitalizing all letters* and generated `boot.scr` with `mkimage`. Then on `/etc/fstab` I replaced all `/dev/mmcblkXpX` with `UUID=xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx`.

Everything is working. Now I can swap cards between slots and everything stays fine, no more kernel panics.

(Note the different between UUID and PARTUUID. `bklid` outputs both of them. PARTUUID is stored in the partition table and is the only one available at boot time for U-Boot to use. UUID is stored inside ext partitions and should be the one used in fstab.)

As @zador-blood-stained pointed out, this can be automatically accomplished in the build scripts:

> run lsblk from build script inside mounted image, parse its output and create boot.cmd and boot.scr on the fly

Also, modify `/etc/fstab` accordingly. Partition UUID won't change from then on. When you install Armbian you do a `dd` to the sdcard. Thus partition UUID is kept the same. Later, on first boot, it is just resized, which also keeps UUID intact.

Can you give it a try @igorpecovnik ?

igorpecovnik, 2015-10-24:
OK, I'll try to implement it ASAP. Generating is an easy part ... I already add this some time age and then remove. I forgot what was the reason ...  I also need to fix SATA / USB / NAND install scripts and I have to test odd uboot / kernel scenarios if everything works as expected.

golfromeo-fr, 2016-10-02:
I have seen later stages for this opened "issue", PARTUUID implies GPT instead of MSDOS partition

at the same time, someone was writing about LABEL (working on vanilla uboot/kernel) instead which might be easier but requires both (UUID/LABEL, I guess) a kernel with ramboot (initrd) to recognize the partitions (& of course a "rootwait" on the kernel commandline)

I added "LABEL=rootfs" for ext4 partition in my PL for XU4 which can be reused. I have no idea if it is working for 3.x kernels.

zador-blood-stained, 2016-10-02:
> PARTUUID implies GPT instead of MSDOS partition

No, they just have a different format:

```
# this is a disk with DOS partition scheme
/dev/sda1: (...) PARTUUID="0004cde8-01"
# this is a disk with GPT partition scheme
/dev/sdb1: (...) PARTUUID="dcc8df90-01c7-4624-99f9-98a8ce11ea9f"
```

> at the same time, someone was writing about LABEL (working on vanilla uboot/kernel) instead which might be easier but requires both (UUID/LABEL, I guess) a kernel with ramboot (initrd) to recognize the partitions (& of course a "rootwait" on the kernel commandline)

PARTUUID does not require initrd, UUID (I mean filesystem UUID, not partition UUID) and LABEL - won't work without initrd

golfromeo-fr, 2016-10-02:
@zador-blood-stained 
ok, I trust you for PARTUUID & no initrd.But can GPT be a universal replacement for old MSDOS partitioning (which only handle UUID&Label through the Ext(/fat32?) filesystem?
https://wiki.archlinux.org/index.php/Persistent_block_device_naming#by-partuuid

zador-blood-stained, 2016-10-02:
No, we can't use GPT for bootable SD and eMMC since partition table overlaps SPL and u-boot for most devices.

golfromeo-fr, 2016-10-02:
ok, so LABEL/UUID is the only options left :(

zador-blood-stained, 2016-10-15:
Fixed with 1946493 for sunxi boards, will be slowly tested and implemented for other.

closed 2016-10-15 by zador-blood-stained