armbian/build #120: Errors building on (x)ubuntu 15.10.

mpmc, 2015-10-19:
OS:
Ubuntu 15.10 32 Bit.

Issue.
Random errors at the end of the compile. compile.sh __not__ altered.
```
[ o.k. ] Install board specific applications. [ bananapi ]
[ o.k. ] Fingerprinting. [ Armbian 4.5 Bananapi Debian jessie next Linux 4.2.3 ]
/home/mark/build/lib/scripts/sunxi-temp/sunxi_tp_temp.c:16:7: warning: type defaults to ‘int’ in declaration of ‘SoCTempAdjustment’ [-Wimplicit-int]
 const SoCTempAdjustment = 1447 ;
       ^
[ o.k. ] Creating boot scripts [ bananapi ]
[ o.k. ] Possible after install. 
[ o.k. ] Writing boot loader [ /dev/loop0 ]
[ o.k. ] Shrink image last partition to [ minimum ]
dumpe2fs 1.42.12 (29-Aug-2014)
1: unknown command
Re-reading the partition table failed.: Invalid argument
[ o.k. ] Create and sign [ Armbian_4.5_Bananapi_Debian_jessie_4.2.3.zip ]
[ o.k. ] Runtime [ 122 min ]
```
__Some background: Warning may be a bit long-winded__

This is the third build I've attempted :(

I originally ran the script on my desktop running Xubuntu 14.04 64 bit which unfortunately brought down the whole system, I'm not sure what caused it, the machine just completely froze near the end.

I then used virtualbox with Lubuntu 14.04 32 bit as guest on the same machine it didn't hang this time & it built fine (as far as I could tell) unfortunately my Banana Pi would not boot, it gets stuck on "Starting kernel", I think there is an issue with my card reader used on the machine I know this because I tried burning an image for the Orange Pi PC using the same reader & that failed too. However I burnt the same Orange Pi image on my laptop and that ran fine.

As I knew the card reader on my spare laptop works fine I decided to wipe W10 off it & use it to build the images instead, the above log is from that machine (Fujitsu Lifebook AH530 - P6200 - 4GB).

I haven't even tried the image as I'm not even sure it would boot with the above errors.

Apologies for the long post but I thought I'd best give as much background info as possible, if you are already aware of these errors, please feel free to close this issue.

igorpecovnik, 2015-10-24:
I made a run on Xubuntu 15.10 but it has incompatible compiler so it won't go. This require additional source patching which is pointless. Use Ubuntu 14.04 LTS based distro as recommended or manually adjust to proper compiler.

Orangepi PC / Plus (H3 based) is not supported.

mpmc, 2015-10-24:
```I made a run on Xubuntu 15.10 but it has incompatible compiler so it won't go.```
Fair enough but what happens when 16.04 gets released? :(

```Orangepi PC / Plus (H3 based) is not supported.```
Sorry I should have been a bit clearer, I know Armbian doesn't support the Orange Pi PC I was merely stating the fact I think the card reader I used is to blame as the OPi PC image also failed to boot on the OPi PC. The image I was referring to was the one offered on orangepi.org.

I guess my only solution for the moment is to install 14.04.

igorpecovnik, 2015-10-24:
Successful cross compilation over many different kernel sources with one compiler is a combination of pure luck and a hard work. If you want to end up with a bootable image stick to recommendations.

This project needs to be supported in order to work properly so it will be fixed on the way for future (LTS based) hosts.

closed 2015-10-24 by igorpecovnik