armbian/build #122: Compile script mangles install of Xubuntu 14.04.3

mpmc, 2015-10-29:
Having re-installed Xubuntu 14.04.3 to run the scripts which appear to have ran fine & built an image however it would appear that the script(s) do something to the current users .config files. I'm still able to login but nothing loads, no lxpanels etc. I found this out when rebooting the machine.

What exactly that is I cannot be sure, what I do know is that it does not affect Lubuntu 14.04.3.

Thankfully I only ran it on my spare machine & not my main one, I suppose this is why it's recommended that the script be ran in a VM or chroot.

But surely the script shouldn't mangle a clean install like this?

igorpecovnik, 2015-10-29:
Hmm, it shouldn't. Which file was altered? Location, where did you run the script from ... did you get any error, there is a debug directory now, check if there is something suspicious in logs. 

I will add some extra check points to make sure it doesn't do any wrong actions.

mpmc, 2015-10-29:
The only thing I could find was this in xession-errors.

````
mktemp: failed to create file via template ‘/tmp/config-err-XXXXXX’: Permission denied
/usr/sbin/lightdm-session: line 29: : No such file or directory
cat: : No such file or directory
truncate: cannot open ‘’ for writing: No such file or directory
/usr/sbin/lightdm-session: line 29: : No such file or directory
cat: : No such file or directory
truncate: cannot open ‘’ for writing: No such file or directory
openConnection: connect: No such file or directory
cannot connect to brltty at :0
Script for none started at run_im.
Script for auto started at run_im.
Script for default started at run_im.
init: logrotate main process (1771) terminated with status 1
init: Disconnected from notified D-Bus bus
```
/usr/sbin/lightdm-session exists..

igorpecovnik, 2015-10-29:
I am using /tmp and tmp file "results" in the current run directory and that's the only relation that I can come up with now. I'll try to replicate.

mpmc, 2015-10-29:
Its, Xubuntu 14.04.3 x64 - There were no errors during the script run & the image built fine however it didn't boot (yours from armbian.com did though!).

I haven't wiped the machine yet so if you need any more files/info I can give them to you :)

gyenchev, 2015-11-03:
Same problem on linux mint 17.2 it seems to be that something is changing the permitions in the system.

igorpecovnik, 2015-11-04:
It's near to impossible to have a full working script across this many systems  / generations. Even if / when I manage to fix the scripting part to be working everywhere I would need to fix kernel code to be compilable with one compiler (insane) or use many compilers together with further host adjustments (insane). And we all want that it just works ;) Well, it works fine on Ubuntu Trusty. So far.

closed 2015-11-04 by igorpecovnik