armbian/build #123: readme prerequisities

serus22, 2015-11-11:
add prerequisities 
(for me working after $ sudo apt-get install gawk wget git-core diffstat unzip texinfo gcc-multilib build-essential chrpath socat libsdl1.2-dev xterm same as for yocto)

igorpecovnik, 2015-11-11:
I am downloading dependencies here:
https://github.com/igorpecovnik/lib/blob/second/main.sh#L234

Is something missing and on what host?

serus22, 2015-11-11:
sorry then
but it wont work before i install them manually 
always it stops with error: "kernel wasnt build, please check configuration..."

(ubuntu 14.04 in virtualbox)

btw, can u help me, when i should put patch file if i want to enable can bus for udoo-quad
http://pastebin.com/BF2SeWHa
or just i must change it manually, 
im trying build this for a 3 days without success :( lvl noob

closed 2015-11-11 by serus22

igorpecovnik, 2015-11-11:
I'll check regarding dependencies.

This can be (is) a problem in kernel source. Udoo kernel is currently hard to compile since it's note done yet.