armbian/build #125: Please add lm75 kernel module

gknauf, 2015-11-18:
root@bananapi:~# uname -a
Linux bananapi 3.4.109-sun7i #4 SMP PREEMPT Sun Oct 11 14:32:15 CEST 2015 armv7l GNU/Linux
root@bananapi:~# zgrep 'LM75' /proc/config.gz
# CONFIG_SENSORS_LM75 is not set
would be great if you could include the lm75 kernel module with your next builds;
also if possible can you perhaps just build it and attach here? Or am I forced to build it myself?
There are some cheap lm75a-based modules available, f.e.:
http://www.ebay.com/itm/331655700684
http://www.ebay.com/itm/400974994990

igorpecovnik, 2015-11-21:
Added whole section. Will be in next build.

closed 2015-11-21 by igorpecovnik

gknauf, 2015-11-21:
Thanks Igor! And of course also thanks for all your work on making Armbian available!