armbian/build #126: qemu fatal error

blindpet, 2015-11-20:
Is this just a fluke or is there a bad patch?
```
[ o.k. ] Extracting jessie from cache [ 0 days old ]
[ o.k. ] Install kernel [ linux-image-next-sunxi_4.6_armhf.deb ]
[ o.k. ] Install u-boot [ linux-u-boot-next-sunxi_4.6_armhf.deb ]
[ o.k. ] Install headers [ linux-headers-next-sunxi_4.6_armhf.deb ]
[ o.k. ] Install firmware [ linux-firmware-image-next-sunxi_4.6_armhf.deb ]
[ o.k. ] Install DTB [ linux-dtb-next-sunxi_4.6_armhf.deb ]
[ o.k. ] Fixing release custom applications. [ jessie ]
[ o.k. ] Install board specific applications. [ bananapipro ]
[ o.k. ] Fingerprinting. [ Armbian 4.6 Bananapipro Debian jessie next Linux 4.2.6 ]
[ o.k. ] Creating boot scripts [ bananapipro ]
[ o.k. ] Possible after install. [ git clone https://github.com/blindpet/MediaServerInstaller HTPCGuides ]
Cloning into 'HTPCGuides'...
remote: Counting objects: 1167, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 1167 (delta 2), reused 0 (delta 0), pack-reused 1161
Receiving objects: 100% (1167/1167), 920.12 KiB | 583.00 KiB/s, done.
/build/qemu-Ee59aw/qemu-2.0.0+dfsg/tcg/optimize.c:362: tcg fatal error
/build/qemu-Ee59aw/qemu-2.0.0+dfsg/tcg/optimize.c:362: tcg fatal error
```

igorpecovnik, 2015-11-20:
What is your host? I haven't notice such problem.

blindpet, 2015-11-20:
On 14.04 Ubuntu 64-bit, maybe the VM is borked, will try again and re-open if it happens on a fresh VM.

closed 2015-11-20 by blindpet