armbian/build #129: libvdpau-sunxi second fix

scpcom, 2015-12-06:
Build host:
Ubuntu 14.04.3 LTS amd64

Build parameters used:
BOARD="bananapi"						# bananapi
BRANCH="default"						# default=3.4.x, mainline=next
RELEASE="jessie"                                   		# jessie, jessie or trusty

Thank you for this great image build script. This makes it easy to get all the things together to get all requireds parts to make the banana pi work with full hardware support.

Adding libvdpau-sunxi does still not work, here are the patches that fixed it:

```shell
# cd to local repo     
sudo sed -i 's|"git checkout 906c36ed45ceb53fecd5fc72e821c11849eeb1a3"|"cd /tmp/libvdpau-sunxi \&\& git checkout 906c36ed45ceb53fecd5fc72e821c11849eeb1a3"|g' lib/desktop.sh
# correct path for vdpau in Debian/Ubuntu is /usr/lib/arm-linux-gnueabihf
sudo sed -i 's|d=$DEST/cache/sdcard/usr/lib/vdpau|d=$DEST/cache/sdcard/usr/lib/arm-linux-gnueabihf/vdpau|g' lib/desktop.sh
# correct symlink
sudo sed -i 's|ln -s $d/libvdpau_sunxi.so $d/libvdpau_sunxi.so.1|ln -s libvdpau_sunxi.so.1 $d/libvdpau_sunxi.so|g' lib/desktop.sh
```

Once the image is built, you can boot to it and do the following tests:

```shell
# test egl
sudo apt-get --no-install-recommends install -y \
  pulseaudio-utils \
  qjackctl
qjackctl

# test vdpau
sudo apt-get install -y mpv
mpv --vo=vdpau --hwdec=vdpau --framedrop=vo --hwdec-codecs=all https://www.youtube.com/watch?v=FGBhQbmPwH8
```

I also had some issue in shrink phase:

```shell
# sometimes parted fails reading partition table because partition end sector is higher than total count of sectors
sudo sed -i s/'ROOTSTART"s" -1s'/'ROOTSTART"s" -255s'/g lib/deboostrap.sh
```

igorpecovnik, 2015-12-27:
Thanks!

https://github.com/igorpecovnik/lib/commit/02d6146f11d731ea2835780d06e77502612a62e5

Fixed

There is some sound delay - any ideas how to fix that?

scpcom, 2015-12-30:
I did not find a general solution for video/audio sync.
This is the reason why I use the option "--framedrop=vo" in mpv.
With this option it is good enough for me to watch streams of TV shows in HD/Fullscreen.

igorpecovnik, 2016-01-26:
No information regarding MALI / CedarX in 4.x ... I think it's far away.

closed 2016-01-26 by igorpecovnik