armbian/build #135: Why not to add lirc_gpio support to sunxi3.4 kernel for A20 chips?

zerkalica, 2016-01-02:
I try to make ir sender from my banana pro, but lirc can't send anything to gpio ports without lirc_gpio kernel driver.

You use mainline kernel with banana-related patches https://github.com/igorpecovnik/lib/blob/master/configuration.sh#L281, without lirc_gpio.
But alternative dan-and kernels has lir_gpio driver: https://github.com/dan-and/linux-sunxi/blob/dan-3.4.current/drivers/staging/media/lirc/lirc_gpio.c
And you fork: 
Originally from https://github.com/matzrh/sunxi-lirc/blob/master/lirc_gpio.c

Why not to add lirc_gpio support to sunxi3.4 kernel for A20 chips?
Why do you use mainline kernel for bananapi/pro, no alternative kernels, which looks more feature-rich?

igorpecovnik, 2016-01-02:
I know, I add those patches to that kernel source ;) but I forgot the reason why those patches are not brought to the kernel I use now by default. Reason of switching back to default community kernel is because it's working with A10 and A20, while dan-and is crippled for A10. I can add those two patches back if they work o.k.? I never came that far to actually test this functionality.

Mainline kernel has it's own key benefits even it's not that feature rich.

zerkalica, 2016-01-04:
GPIO lirc transmitter works fine:
Working configs for banana pro:
https://gist.github.com/zerkalica/f1c3a0a889cd974c495d

1. Apply patch from https://github.com/matzrh/sunxi-lirc
2. Add lirc_gpio to /etc/modules
3. Get lirc configs from my gist
4. Flash led: irsend -d /dev/lircd1 SEND_ONCE roku KEY_UP
 
Need separate configs for two lircd instances. Armbian /etc/init.d/lirc is buggy - TRANSMITTER_LIRCD_CONF unused. I have patched /etc/init.d/lirc: added TRANSMITTER_LIRCD_CONF and TRANSMITTER_PIN variables.

In this table use sys row for banana pro to find sys pin number for given hardware pin. Example: hardware pin 3 PB21 (WTI2_SDA) is sys pin 2.

http://wiki.lemaker.org/How_to_control_the_IO_on_the_SBC_boards#Appendix:_40_Pins_GPIO_Mapping_Table_for_Banana_Pro_and_LeMaker_Guitar

zerkalica, 2016-01-05:
One note, sunxi-ir driver is buggy - supports only NEC protocol. irrecord can't record on my device.
But irrecord works with sunxi_lirc . Please, add lirc_gpio and sunxi_lirc to you kernel patches.

http://forum.lemaker.org/thread-6657-1-1.html

igorpecovnik, 2016-01-26:
https://github.com/igorpecovnik/lib/commit/e02e08a366f0bab2e2da46dc8e1900ef3125c82f

I added patches to the A10/20 kernel and sucesfully conduct IRSEND test. I am only a bit confused about the receive part. We don't have devinput with this driver, right? Does it make sense to retain default driver?

zerkalica, 2016-01-26:
With sunxi_lirc use /dev/lirc1 or lirc0 instead of devinput.

But sunxi_lirc is buggy too =) mode2 shows numbers if key pressed on remote. irrecord works - write dots, if key pressed, but can't produce working config. And lirc daemon doesn't work with sunxi_lirc: i  think, it don't understand protocol, which used in sunxi_lirc.

Anhother bug: when playing sound from onboard or external usb sound card - and data sending to lirc_gpio - pulseaudio is shut up. Helps this: pasuspender -- irsend ...

Overall:
sunxi_ir - lirc can receive from devinput, but irrecord can't record
sunxi_lirc - irrecord can record, but lirc can't receive from lirc0
lirc_gpio - irsend can send to gpio, but sending affects some devices, like soundcards (think, this is kernel or driver bug, problem with timers, etc)

igorpecovnik, 2016-01-26:
Hmm, nice ;) OK, let's include the drivers into the build and if someone needs this functionality, he could relate to your findings, use a proper combination, not waste time and perhaps provide a fix for this.

dobril-stanga, 2016-02-23:
There are no /dev/lirc0 or /dev/lirc1 with module sunxi_ir
sunxi_lirc create  /dev/lirc0  , but irrecord cannot record the KEYs:

Press RETURN now to start recording.
................................................................................
Found gap: 16275
Please keep on pressing buttons like described above.
................................................................................
Space/pulse encoded remote control found.
Signal length is 67.
No header data.
Found trail pulse: 537
Found repeat code: 8912 2237
Found repeat gap: 16275
Signals are space encoded.
Signal length is 33
Now enter the names for the buttons.

Please enter the name for the next button (press <ENTER> to finish recording)
KEY_EXIT

Now hold down button "KEY_EXIT".
Something went wrong. Please try again. (9 retries left)

damien7851, 2016-02-25:
Hello,
I recently finished to port ir driver of the mainline to the legacy.
It decodes all protocol supported by rc-core.
For now I could check that returns the scancode for two remote: a yamaha in NEC and Samsung.
I just found a bug to confirm: a kernel panic with LIRC protocol and mode2  when very long press of a button.

source code are hosted on github : https://github.com/damien7851/lirc_sunxi_rewrite.git branch rc.

I'm very happy to receive code review and pull request.

dobril-stanga, 2016-02-25:
Can you tell me  how I can build lirc_gpio module in kernel 4.3
You build the module from lirc source ?

damien7851, 2016-03-21:
Hi,
I haven't compliled my driver with kernel 4.3 but with 3.4 kernel. It is not useful to compile lirc. You just compiled my driver with the sources of the kernel. After you must add the module in /etc/modules.
this activate rc-core fonction and you can use this driver as standard driver following common doc of lirc.
(i-e with keytable software)

zador-blood-stained, 2017-10-28:
Obsolete since kernel 3.4.x is EOL, if anybody wants to add/improve anything in this kernel a pull request can be used.

closed 2017-10-28 by zador-blood-stained