armbian/build #138: Would you like to have a build server?

faddat, 2016-01-05:
Hey,

At this stage in the game my code contribution might not do much .  That said, I'm wondering if you have any sort of a continuous build system going to provide you with feedback?  If not, let me know and I'll be glad to help get one set up, and we'd be glad to host it :).

igorpecovnik, 2016-01-26:
Thanks for your offer. Currently we are fine with hardware. The reason why auto building is not done is lack of human resources to ajdust / fix / update scripts. Not long time ago we fixed some fundamental problems. It's on the plan for some time and we are close to live action ;)

closed 2016-01-26 by igorpecovnik

faddat, 2016-01-27:
Let me know if there are any specific tasks in that dept that you'd like a
hand with.  :).

Jacob Gadikian
E-mail: faddat@gmail.com
SKYPE: faddat
Phone/SMS: +84 167 789 6421