armbian/build #139: alsa

norrest, 2016-01-12:
Hi, I used  Cubieboard a10 and 
uname -a
Linux cubieboard 4.2.3-sunxi #2 SMP Sun Oct 11 14:12:18 CEST 2015 armv7l GNU/Linux
But i do not find alsa ))
cat /proc/asound/version
cat: /proc/asound/version: No such file or directory

igorpecovnik, 2016-01-12:
Audio come with kernel 4.4

norrest, 2016-01-13:
Thank you! 
And if I can define another setting in the kernel? 4.2.3 Can not really suitable?

igorpecovnik, 2016-01-13:
select DEV and it will download 4.4 .. Its working, tested on Cubie a10 today.

norrest, 2016-01-14:
Yes! 
 cat /proc/asound/version
Advanced Linux Sound Architecture Driver Version k4.4.0-sunxi.
THANK YOU!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

norrest, 2016-01-14:
I must to reopen issue! ((
I found on kernel.org 
https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.4.tar.xz
unpack, and in /include/uapi/sound/asound.h  
#define	SNDRV_PCM_FORMAT_DSD_U8		((__force snd_pcm_format_t) 48) /* DSD, 1-byte samples DSD (x8) */
#define	SNDRV_PCM_FORMAT_DSD_U16_LE	((__force snd_pcm_format_t) 49) /* DSD, 2-byte samples DSD (x16), little endian */
#define	SNDRV_PCM_FORMAT_DSD_U32_LE	((__force snd_pcm_format_t) 50) /* DSD, 4-byte samples DSD (x32), little endian */
#define	SNDRV_PCM_FORMAT_DSD_U16_BE	((__force snd_pcm_format_t) 51) /* DSD, 2-byte samples DSD (x16), big endian */
#define	SNDRV_PCM_FORMAT_DSD_U32_BE	((__force snd_pcm_format_t) 52) /* DSD, 4-byte samples DSD (x32), big endian */
#define	SNDRV_PCM_FORMAT_LAST		SNDRV_PCM_FORMAT_DSD_U32_BE

but in installed ARMBIAN last records is not present

/usr/include/sound
cat asound.h | grep DSD
#define SNDRV_PCM_FORMAT_DSD_U8         ((snd_pcm_format_t) 48) /* DSD, 1-byte samples DSD (x8) */
#define SNDRV_PCM_FORMAT_DSD_U16_LE     ((snd_pcm_format_t) 49) /* DSD, 2-byte samples DSD (x16), little endian */
#define SNDRV_PCM_FORMAT_LAST           SNDRV_PCM_FORMAT_DSD_U16_LE

It is for me critical, because i try to work with MPD-DSD

https://github.com/lintweaker/xmos-native-dsd/tree/master/SRPMS/patches/kernel 

which is already working in the new kernels, without the use of these patches!
The last kernel already included them in yourself!

Thank YOU!

norrest, 2016-01-15:
I post this

norrest, 2016-01-15:
Maybe i use wrong repository? 
I delete linux-libc-dev:armhf                 3.16.7-ckt20-1+deb8u2 but it  again 
dpkg -l 'linux-*'
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                 Version                 Architecture            Description
+++-====================================-=======================-=======================-=============================================================================
un  linux-doc-4.4.0                      <none>                  <none>                  (no description available)
ii  linux-dtb-dev-sunxi                  4.83                    armhf                   Linux DTB, version 4.4.0-sunxi
ii  linux-firmware-image-dev-sunxi       4.83                    armhf                   Linux kernel firmware, version 4.4.0-sunxi
un  linux-headers                        <none>                  <none>                  (no description available)
un  linux-headers-3.0                    <none>                  <none>                  (no description available)
ii  linux-headers-4.4.0-040400           4.4.0-040400.2016011019 all                     Header files related to Linux kernel version 4.4.0
ii  linux-headers-4.4.0-040400-generic   4.4.0-040400.2016011019 armhf                   Linux kernel headers for version 4.4.0 on 64 bit x86 SMP
ii  linux-headers-dev-sunxi              4.83                    armhf                   Linux kernel headers for 4.4.0-sunxi on armhf
un  linux-image                          <none>                  <none>                  (no description available)
un  linux-image-2.6                      <none>                  <none>                  (no description available)
un  linux-image-3.0                      <none>                  <none>                  (no description available)
ii  linux-image-4.4.0-040400-generic     4.4.0-040400.2016011019 armhf                   Linux kernel image for version 4.4.0 on 64 bit x86 SMP
ii  linux-image-dev-sunxi                4.83                    armhf                   Linux kernel, version 4.4.0-sunxi
un  linux-initramfs-tool                 <none>                  <none>                  (no description available)
ii  linux-jessie-root-cubieboard         4.83                    armhf                   Root file system tweaks for cubieboard
un  linux-kernel-headers                 <none>                  <none>                  (no description available)
un  linux-kernel-log-daemon              <none>                  <none>                  (no description available)
ii  linux-libc-dev:armhf                 3.16.7-ckt20-1+deb8u2   armhf                   Linux support headers for userspace development
un  linux-modules-4.4.0-sunxi            <none>                  <none>                  (no description available)
un  linux-source-4.4.0                   <none>                  <none>                  (no description available)
un  linux-tools                          <none>                  <none>                  (no description available)

norrest, 2016-01-15:
Aslo i try to install  4.4.0-040400.2016011019 deb (kernel + header) itis do not help (

igorpecovnik, 2016-01-26:
I have no detail knowledge on this issue. Sound function just came to mainline so things are probably not yet developed to full degree?

closed 2016-02-15 by norrest