armbian/build #146: [REQUEST]

Tom-Neverwinter, 2016-02-15:
can you release in a more windows friendly format. Imagewriter is annoying to work with and can be hit or miss as it is very finicky of memory card readers (yeah its finicky, my reader writer has been doing this forever and has no known issues still). Win32diskimager gives better results and I can modify the data before transcribing to a memory card.

igorpecovnik, 2016-02-15:
It's a known problem. 
https://github.com/igorpecovnik/lib/blob/master/documentation/user-faq.md#how-to-prepare-sd-card

I added a note to docs that you better use https://rufus.akeo.ie/ which is also not perfect but it works everywhere.

Tom-Neverwinter, 2016-02-15:
That helped get the image on the micro sd card, now I just need to sit down and figure out how to setup my 3.5 touch panel

closed 2016-02-16 by igorpecovnik