armbian/build #150: Orange Pi PC H3 only one USB port can be used

bandesznet, 2016-02-16:
I have built and tested: Armbian_5.00_Orangepipc_Ubuntu_trusty_3.4.110.zip
Build time: 2016-02-15 11:51 GMT+1
The USB hub bug already fixed by loboris in the legacy kernel.

igorpecovnik, 2016-02-16:
Do you know more on the issue? Was it a kernel or FEX board configuration?

bandesznet, 2016-02-16:
I don't know. The only thing I know is that loboris fixed it in this post: http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=471&page=1#pid3822
There is no comment or description how can be fixed, only a download link to the pathed files. He makes uimage and script.bin together. It is published on his repo: https://github.com/loboris/OrangePI-Kernel

WereCatf, 2016-02-19:
I just tested and, indeed, only one of the USB-ports works as it should. The other two work fine if you connect a high-speed USB-device on them, but as soon as you connect a low-speed device on them they stop working and won't start to work again even if you went and plugged a high-speed device in -- only a reboot fixes that.

When connecting a low-speed device to the one port that works it's taken over by sunxi-ohci, but connecting to one of the non-working ports results in "ehci_irq: highspeed device connect" followed by "ehci_irq: highspeed device disconnect" in dmesg.

I took a look at the settings the kernels were compiled with and therein lies the culprit: Loboris - kernel has CONFIG_USB_SUNXI_OHCI0, CONFIG_USB_SUNXI_OHCI1, CONFIG_USB_SUNXI_OHCI2 and CONFIG_USB_SUNXI_OHCI3 set whereas this Armbian - kernel doesn't. Turning those on should make the USB-ports functioning again.

bandesznet, 2016-02-26:
Great! Thanks!

closed 2016-02-26 by bandesznet