armbian/build #156: sun8i default config, missing modules and gpio sysfs-interface?

WereCatf, 2016-02-19:
Is there some good rationale behind not compiling all the drivers the kernel offers as modules? There are a whole boatload of stuff there and even very basic stuff like USB-printers aren't enabled in the config and I do have to wonder why. As things stand you're pretty much forced to compile your own kernel to do much of anything useful with Armbian on, say. Orange Pi - devices.

Also, on a similar vein, I wonder why support for the gpio sysfs-interface has been left out? Including it wouldn't increase kernel-size, it wouldn't break anything, and the included gpio_sunxi - module doesn't offer the same functionality, meaning that e.g. one cannot use WiringOP's interrupt-support without compiling a whole kernel of their own.

igorpecovnik, 2016-02-19:
We were focusing on other things, no particular reason.
http://forum.armbian.com/index.php/topic/617-wip-support-for-the-upcoming-orange-pi-one/

It's planned to provide as much as possible - similar to sun7i config. Any help is appreciated.

WereCatf, 2016-02-19:
Alright then, good to know. I just thought that maybe there is a good reason for it, but I'm just unaware of it. I've got an OPi PC, so I'll probably be submitting a patch here and there related to it.

Alas, I suppose that "fixes" this issue and I'm closing it, then!

closed 2016-02-19 by WereCatf