armbian/build #163: Orange Pi PC 3.4 - kernel doesn't reboot

WereCatf, 2016-02-20:
No idea yet why, but reboot doesn't actually reboot -- it shuts down. Only power-recycle brings the board back up again.

igorpecovnik, 2016-02-20:
It's working fine with my Opi Plus. Reboot, reboots ;) ... but yes, we need to find out why this happen on some boards and fix it.

WereCatf, 2016-02-20:
I'm not exactly swimming in cash, so I can't really donate any boards. At least not yet :/

I'll have to try with loboris fex - files or something later on and see if the board reboots fine with those -- if not, then it's something else that's causing it.

WereCatf, 2016-02-20:
It is the fex - file: the OPi PC reboots fine with Loboris - one, but not Armbian - one

ThomasKaiser, 2016-02-20:
My Orange Pi One was also stuck at rebooting but now it reboots fine (latest kernel and fex fixes)

martinayotte, 2016-02-20:
On my OrangePi-PC, I've tried the Igor's 5.0.3 image, and the "reboot" simply hangs without doing any thing else. It even occurs during the "firstrun" script, I had to recycle power to let other phases going.

WereCatf, 2016-02-23:
Pull request #174 fixes this, closing.

closed 2016-02-23 by WereCatf