armbian/build #170: Display-output broken

WereCatf, 2016-02-21:
Some recent change has broken display-output on the OPi PC. I have an HDMI TV connected to it and in 5.01 it worked without me having to adjust anything. Now I can't seem to make it work. I just tried with the Loboris fex - file and it didn't do squat, so it's something else that's caused this -- the display is initialized, the TV says it's 1080p as it should be, but content stays black, no matter what.

igorpecovnik, 2016-02-21:
I also experience no HDMI and later figured out that this is causing the problem:
https://github.com/igorpecovnik/lib/commit/65830b7cdce91a40281ee4f20d45e8c76e6aa932

The problem must be elsewhere since this particular switch should not affect. I am having OPi+ and I have HDMI out with latest images from download section (5.03).

WereCatf, 2016-02-21:
I was fiddling with the kernel while I was testing 5.01 and at that time turning that flag on or off didn't break display. Something since then has changed, though I can't fathom why changing that flag would affect HDMI - display o_o

WereCatf, 2016-02-21:
Yeah, I can confirm: turning optimize for size on fixes display, turning it off breaks it -- just tested it.

ThomasKaiser, 2016-02-25:
@WereCatf if you want to give it a try again I would suspect it's [patch 0006](https://github.com/igorpecovnik/lib/commits/master/patch/kernel/sun8i-default/0006-h3-videomode-cmdline.patch). This patch would allow to set display resolution through kernel cmd_line but in an incompatible way compared to all other sunxi boards (there 'disp.screen0_output_mode=1920x1080p60' with out patch 'disp.screen0_output_mode=10'). So I wouldn't mind if we disable this patch for now (maybe someone ports all the other sun7i display stuff like correct EDID initialisation and so on) and try whether we can go with disabled 'optimize for size'?

WereCatf, 2016-02-25:
No, doesn't look like it's that. I did try without, but there was no change. I don't think optimize for size being broken is a major issue at the moment as it doesn't break anything else and considering how long it always takes for me to compile everything and all I don't really have much interest in fiddling with it right now. I'm just gonna leave this issue open in case someone else gets bored and wants to dig into this.

ThomasKaiser, 2016-02-25:
@WereCatf Thx for the try, then I will check 0007 instead the next time I rebuild things.

ThomasKaiser, 2016-02-25:
Disabling [0007-fbdev-fixes.patch](https://github.com/igorpecovnik/lib/blob/master/patch/kernel/sun8i-default/0007-fbdev-fixes.patch) didn't also work. HDMI output is enabled but screen is still blank when _#CONFIG_CC_OPTIMIZE_FOR_SIZE is not set_

closed 2016-06-05 by zador-blood-stained